qca/open-ath9k-htc-firmware #72: ath: phy82: Failed to wakeup in 500us

yipperr, 2015-02-15:
hello i am using a alpha awus036nha in ubuntu with 3.18 kernel causes a lot of disconnects where the whole driver reloads itself and the usb layer is deinitialized 

here is the dmesg

[134330.226799] ath: phy82: Failed to wakeup in 500us
[134330.337247] ath: phy82: Failed to wakeup in 500us
[134330.553857] cfg80211: Calling CRDA to update world regulatory domain
[134330.557245] cfg80211: World regulatory domain updated:
[134330.557249] cfg80211:  DFS Master region: unset
[134330.557250] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[134330.557252] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[134330.557254] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[134330.557255] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm), (N/A)
[134330.557256] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[134330.557257] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[134330.621383] usb 1-1: ath9k_htc: USB layer deinitialized
[134330.861057] usb 1-1: new high-speed USB device number 87 using xhci_hcd
[134331.008352] usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271
[134331.008363] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[134331.008369] usb 1-1: Product: UB91C
[134331.008373] usb 1-1: Manufacturer: ATHEROS
[134331.008377] usb 1-1: SerialNumber: 12345
[134331.010014] usb 1-1: ath9k_htc: Firmware htc_9271.fw requested
[134331.472999] usb 1-1: ath9k_htc: Transferred FW: htc_9271.fw, size: 51272
[134331.708841] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[134331.938832] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.3
[134331.938843] ath: EEPROM regdomain: 0x833a
[134331.938846] ath: EEPROM indicates we should expect a country code
[134331.938851] ath: doing EEPROM country->regdmn map search
[134331.938855] ath: country maps to regdmn code: 0x37
[134331.938859] ath: Country alpha2 being used: GB
[134331.938861] ath: Regpair used: 0x37
[134331.943193] ieee80211 phy83: Atheros AR9271 Rev:1
[134331.943229] cfg80211: Calling CRDA for country: GB

olerem, 2015-02-15:
please try latest firmware
https://github.com/olerem/ath9k-htc-firmware-blob

Beside, i have similar adapter. What is your reproduction case?
How do you know that windows driver runs solid, do you have access to the logs provided by this driver? :)

yipperr, 2015-02-15:
i am running the latest wireless backports and the  latest firmware blob also i compiled my own it still acts the same way .

this is the log that happens when the usb device disconnects

[140606.192122] usb 1-1: USB disconnect, device number 92
[140606.199477] wlan2: deauthenticating from xx:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
[140606.212695] ath: phy87: Failed to wakeup in 500us
[140606.323103] ath: phy87: Failed to wakeup in 500us
[140606.539652] cfg80211: Calling CRDA to update world regulatory domain
[140606.542714] cfg80211: World regulatory domain updated:
[140606.542718] cfg80211:  DFS Master region: unset
[140606.542719] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[140606.542722] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[140606.542725] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[140606.542727] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm), (N/A)
[140606.542729] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[140606.542731] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[140606.607097] usb 1-1: ath9k_htc: USB layer deinitialized
[140609.679702] usb 1-1: new high-speed USB device number 93 using xhci_hcd
[140609.827356] usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271
[140609.827366] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[140609.827371] usb 1-1: Product: UB91C
[140609.827375] usb 1-1: Manufacturer: ATHEROS
[140609.827408] usb 1-1: SerialNumber: 12345
[140609.828960] usb 1-1: ath9k_htc: Firmware htc_9271.fw requested
[140610.291181] usb 1-1: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
[140610.541389] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[140610.769990] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[140610.770001] ath: EEPROM regdomain: 0x833a
[140610.770004] ath: EEPROM indicates we should expect a country code
[140610.770009] ath: doing EEPROM country->regdmn map search
[140610.770012] ath: country maps to regdmn code: 0x37
[140610.770024] ath: Country alpha2 being used: GB
[140610.770026] ath: Regpair used: 0x37
[140610.773858] ieee80211 phy88: Atheros AR9271 Rev:1
[140610.773877] cfg80211: Calling CRDA for country: GB
[140610.779323] cfg80211: Regulatory domain changed to country: GB
[140610.779332] cfg80211:  DFS Master region: unset
[140610.779335] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[140610.779342] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[140610.779347] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[140610.779351] cfg80211:   (5250000 KHz - 5330000 KHz @ 40000 KHz), (N/A, 2000 mBm), (0 s)
[140610.779355] cfg80211:   (5490000 KHz - 5710000 KHz @ 40000 KHz), (N/A, 2700 mBm), (0 s)
[140610.779359] cfg80211:   (57240000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[140610.880506] ath9k_htc 1-1:1.0 wlan2: renamed from wlan1
[140610.918467] systemd-udevd[32722]: renamed network interface wlan1 to wlan2
[140611.135072] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[140612.758780] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready

olerem, 2015-02-16:
Uff... lots of questions :), lets do it step by step:
- Please provide complete dmesg.  I need to know more..
- 2mbit is not a lot :D, usually i pass about 90 Mbit to test this adapter. This adapter has some other kind of hiccups by high load. But it will produce other dmesg.
- ath9k_htc is a driver for ar9271 and ar7010 usb adapter, ath9k_htc_fw is a firmware for this adapter. ath9k is a driver for non usb adapter. In any case, ath9k_htc and ath9k share a lot of code. Some times i work on reduction of duplicated code between this two drivers. With other words, don't worry they will not conflict with each other.
- "Failed to wakeup" message comes after usb was disconnected. In this case it is expected message.

Lets try to reduce Interrupt traffic. Seems like it is causing some problems on different HW. Try this:
sudo bash -c "echo none > /sys/class/leds/ath9k_htc-phy*/trigger"
This command will disable LED blinking.

yipperr, 2015-02-16:
by disabling the leds of the wifi card to reduce the interrupt traffic over the usb as u suggested i still get the 

[140606.192122] usb 1-1: USB disconnect, device number 92
[140606.199477] wlan2: deauthenticating from xx:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
[140606.212695] ath: phy87: Failed to wakeup in 500us
[140606.323103] ath: phy87: Failed to wakeup in 500us

followed by the driver reloading itself

[140606.607097] usb 1-1: ath9k_htc: USB layer deinitialized
[140609.679702] usb 1-1: new high-speed USB device number 93 using xhci_hcd
[140609.827356] usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271
[140609.827366] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[140609.827371] usb 1-1: Product: UB91C
[140609.827375] usb 1-1: Manufacturer: ATHEROS
[140609.827408] usb 1-1: SerialNumber: 12345
[140609.828960] usb 1-1: ath9k_htc: Firmware htc_9271.fw requested
[140610.291181] usb 1-1: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
[140610.541389] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[140610.769990] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[140610.770001] ath: EEPROM regdomain: 0x833a
[140610.770004] ath: EEPROM indicates we should expect a country code
[140610.770009] ath: doing EEPROM country->regdmn map search
[140610.770012] ath: country maps to regdmn code: 0x37
[140610.770024] ath: Country alpha2 being used: GB
[140610.770026] ath: Regpair used: 0x37
[140610.773858] ieee80211 phy88: Atheros AR9271 Rev:1
[140610.773877] cfg80211: Calling CRDA for country: GB
[140610.779323] cfg80211: Regulatory domain changed to country: GB
[140610.779332] cfg80211: DFS Master region: unset
[140610.779335] cfg80211: (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[140610.779342] cfg80211: (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[140610.779347] cfg80211: (5170000 KHz - 5250000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[140610.779351] cfg80211: (5250000 KHz - 5330000 KHz @ 40000 KHz), (N/A, 2000 mBm), (0 s)
[140610.779355] cfg80211: (5490000 KHz - 5710000 KHz @ 40000 KHz), (N/A, 2700 mBm), (0 s)
[140610.779359] cfg80211: (57240000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[140610.880506] ath9k_htc 1-1:1.0 wlan2: renamed from wlan1

yipperr, 2015-02-16:
can u specify the particular kernel log your interested the system dmesg log is really huge its been running for some time and the log have accumulated a lot i have already pasted everything that has to concern with the ath9k_htc driver

yipperr, 2015-02-16:
<quote>"Failed to wakeup" message comes after usb was disconnected. In this case it is expected message</quote>

olerem, 2015-02-16:
The driver do not reloading itself. At some point usb layer getting "reset" or "detach" signal from hardware. After this, deinit process is starting, ath9k_htc is not needed any more since usb device is detached, so it is unloaded. Then usb layer get attach signal and the driver is loaded again.
If some job was running and not stopped on driver unload, then this job will fail print some thing like this " Failed to wakeup in".

Forget about pci driver.
First of all recheck the usb cable (some times we get bug reports wich are solved by other cable), test if your adapter get enough power.
Please provide complete! dmesg,, compress it and send to linux ^ rempel-privat ^ de.

yipperr, 2015-02-16:
i restarted the system this is a fresh dmesg after reboot

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.5-031805-generic (kernel@gomeisa) (gcc version 4.6.3 (Ubuntu/Linaro 4.6.3-1ubuntu5) ) #201501292218 SMP Fri Jan 30 03:19:17 UTC 2015
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-3.18.5-031805-generic root=UUID=001de730-e9bf-4b35-9464-840fff728a17 ro quiet splash
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] Disabled fast string operations
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bac8efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bac8f000-0x00000000bacabfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bacac000-0x00000000bacb8fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bacb9000-0x00000000badcefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000badcf000-0x00000000badcffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000badd0000-0x00000000badfafff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000badfb000-0x00000000badfcfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000badfd000-0x00000000badfefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000badff000-0x00000000bae01fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bae02000-0x00000000bae09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bae0a000-0x00000000bae1afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bae1b000-0x00000000baf17fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000baf18000-0x00000000baf1bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000baf1c000-0x00000000baf1efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000baf1f000-0x00000000baf9efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000baf9f000-0x00000000baffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bafff000-0x00000000baffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bb000000-0x00000000bf9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffd80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013e9fefff] usable
[    0.000000] BIOS-e820: [mem 0x000000013e9ff000-0x000000013fdfffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Dell Inc.          Dell System XPS L502X/0NJT03, BIOS A12 09/07/2012
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] AGP: No AGP bridge found
[    0.000000] e820: last_pfn = 0x13e9ff max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0FFC00000 mask FFFC00000 write-protect
[    0.000000]   1 base 000000000 mask F80000000 write-back
[    0.000000]   2 base 080000000 mask FC0000000 write-back
[    0.000000]   3 base 0BC000000 mask FFC000000 uncachable
[    0.000000]   4 base 0BB000000 mask FFF000000 uncachable
[    0.000000]   5 base 100000000 mask FC0000000 write-back
[    0.000000]   6 base 13FE00000 mask FFFE00000 uncachable
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86 PAT enabled: cpu 0, old 0x7040600070406, new 0x7010600070106
[    0.000000] e820: last_pfn = 0xbb000 max_arch_pfn = 0x400000000
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000097000] 97000 size 24576
[... 926 lines cut ...]
[   32.928554] wlan2: associated
[   32.928561] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[   39.378994] bbswitch: module verification failed: signature and/or  required key missing - tainting kernel
[   39.379330] bbswitch: version 0.7
[   39.379334] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   39.379337] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[   39.379345] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20140926/nsarguments-95)
[   39.379477] bbswitch: detected an Optimus _DSM function
[   39.379494] pci 0000:01:00.0: enabling device (0006 -> 0007)
[   39.379522] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   39.381109] bbswitch: disabling discrete graphics
[... cut ...]
[   53.111050] audit: type=1400 audit(1424086630.596:43): apparmor="STATUS" operation="profile_replace" name="third_party" pid=2299 comm="apparmor_parser"
[   53.399482] systemd-logind[1375]: Failed to start unit user@1000.service: Unknown unit: user@1000.service
[   53.399489] systemd-logind[1375]: Failed to start user service: Unknown unit: user@1000.service
[   53.402549] systemd-logind[1375]: New session c2 of user ano.
[   53.402563] systemd-logind[1375]: Linked /tmp/.X11-unix/X0 to /run/user/1000/X11-display.
[   55.240655] init: plymouth-stop pre-start process (2580) terminated with status

yipperr, 2015-02-16:
i have come across many error that might be caused by something trivial like the wpa_supplicant or the network-manger i am asking this because after googling i seem to be the very few people reporting on this error can't find anyone reporting the same for the alpha awus03nha card

nothing in the openwrt forums is also clue because of it being totally difffernt distribution should i look into network-manger doing anything strange ? or is this absurd

olerem, 2015-02-16:
Like i said, i use this adapter and do some stress tests on it. Some times FW may fail and reboot adapter, but in this case with updated kernel you will get some thing like this:
[ 3501.052975] usb 1-2: ath: firmware panic! exccause: 0x0000000d; pc:
0x0090a641; badvaddr: 0x12345678.

Your issue looks more like this:
http://www.linuxforums.org/forum/hardware-peripherals/195227-tl-wn722n-backtrack-5-a.html

olerem, 2015-02-16:
Here is updated list with usb related issues:
https://github.com/qca/open-ath9k-htc-firmware/wiki/usb-related-issues

feedbacks are welcome! :D

Skity, 2015-06-07:
Hi,
I have exactly the same issue with the same adapter than yipperr.
I do it from a Raspberry Pi with the latest atheros drivers.
==================
Linux hostname 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux
==================

Are you still following this issue ? If I can help you resolve it ...

Same symptom than yipperr: Works perfectly (or almost) on Windows. Just have to reduce TX/RX buffer to 128 instead of 256 else it seems to have the same problem on Windows but Microsoft manages better it by "not disconnecting".

Do you know how to change the buffer in linux ?

olerem, 2015-06-07:
i have no idea what "TX/RX buffer" means.

Skity, 2015-06-07:
TX=Transmission
RX=Reception

This is an option from the Windows driver. In the device properties on Windows I can change "Transmit Buffers" and "Receive Buffers"

Skity, 2015-06-08:
I removed the blinked LED by doing 
linux $> echo none > /sys/class/leds/ath9k_htc-phy*/trigger

It seems to be better.
If you don't heard of me it had worked :)

olerem, 2015-06-08:
If you will update kernel and firmware, it should work even better.

Skity, 2015-06-08:
They are both up to date :)

olerem, 2015-06-08:
3.18.11? really?

Skity, 2015-06-08:
Yop -->
root@hostname:~# uname -a
Linux hostname 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux

Skity, 2015-06-08:
$> apt-cache show firmware-atheros
Package: firmware-atheros
Source: firmware-nonfree
Version: 0.36+wheezy.1
Installed-Size: 814
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Architecture: all
Suggests: initramfs-tools, linux-image
Size: 495008
SHA256: 2701f035f46e043ee15d23ed6b7e9860732ee5e58a1c04c7375ff13da4776143
SHA1: ddf5a1bb05e5524faea97cc104abdf3b29c92047
MD5sum: 04d7b3f3d3e1b4367c4850763a0820f0
Description: Binary firmware for Atheros wireless cards
Multi-Arch: foreign
Homepage: http://git.kernel.org/?p=linux/kernel/git/firmware/linux-firmware.git
Description-md5: 5a25e0b2c609d97b2e8ae2eb46c75359
Tag: admin::hardware, hardware::TODO, role::data
Section: non-free/kernel
Priority: optional
Filename: pool/non-free/f/firmware-nonfree/firmware-atheros_0.36+wheezy.1_all.deb
======================================================================
/lib/firmware # ll *9271*
-rw-r--r-- 1 root root 51312 Jan  6  2013 ar9271.fw
-rw-r--r-- 1 root root 51272 Jan  6  2013 htc_9271.fw

===========================================
$> md5sum ar9271.fw htc_9271.fw
2e6f5045ec4c5a42bb93ced242bad0ba  ar9271.fw
76c902360ef8a095133258fd0d287292  htc_9271.fw
================================

If it could help ;-)

olerem, 2015-06-08:
Well.... it is very old kernel and firmware.

Skity, 2015-06-08:
I can't do better I think. I'm on a Raspberry Pi with ARM architecture. Unless I build my own kernel through cross compilation, I can't go to 4.0.5.

Anyway, disabling the Led seems to solved the problem.

yipperr, 2015-06-08:
in my cases the problem din't get solved with turning the led off.. it happens randomly for me where it's most frequent with access point with lower signals and less frequent with access points with higher strength

Skity, 2015-06-08:
@yipperr : Try to maximize the current on your USB ports (since RPI B+) with the following options in config.txt in /boot : 

safe_mode_gpio=4
max_usb_current=1

Skity, 2015-06-08:
I also changed the cable by one with two output (One to the adapter and Two to the RPI to provide more power to the adapter)

yipperr, 2015-06-08:
i tried the usb 3.0 port and 2.0 ports on my laptop all behaves similarly

Skity, 2015-06-08:
@yipperr : not glad it doesn't fix it for you :)
Do you have linux on your laptop ? Did you tried to purge "firmware-atheros" package and re-install it and then upgrade it ?

user113, 2015-08-31:
Advice: English no main language
Hi i just created the account here because i have the same yipperr issue. I have the same device, the alfa card. Im running Debian Jessie stable version.
############################# 
> uname -a
Linux localhost 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u3 (2015-08-04) x86_64 GNU/Linux
#############################
> lsusb -v | egrep "MaxPower|Bus"
Bus 003 Device 006: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n
      (Bus Powered)
    MaxPower              500mA
#############################
Commands i tried to fix.
> echo none > /sys/class/leds/ath9k_htc-phy*/trigger
Commands about power.
> echo "-1" > /sys/bus/usb/devices/usb3/power/autosuspend_delay_ms
> echo "on" > /sys/bus/usb/devices/usb3/power/control
#############################
SYSLOG
kernel: [  279.532867] snd_hda_intel 0000:00:1b.0: IRQ timing workaround is activated for card #1. Suggest a bigger bdl_pos_adj.
kernel: [  323.905675] usb 3-2: USB disconnect, device number 3
kernel: [  323.922344] wlan0: deauthenticating from xx:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
kernel: [  324.025457] ath: phy1: RX failed to go idle in 10 ms RXSM=0x48a1a9c6
kernel: [  324.035503] ath: phy1: Failed to wakeup in 500us  
kernel: [  324.136105] ath: phy1: RX failed to go idle in 10 ms RXSM=0x40b36725
kernel: [  324.146169] ath: phy1: Failed to wakeup in 500us  
kernel: [  324.246797] ath: phy1: RX failed to go idle in 10 ms RXSM=0xc892a9c6
kernel: [  324.256787] ath: phy1: Failed to wakeup in 500us  
kernel: [  324.358372] ath: phy1: RX failed to go idle in 10 ms RXSM=0xc892a9c6
kernel: [  324.381667] cfg80211: Calling CRDA to update world regulatory domain
kernel: [  324.383990] cfg80211: World regulatory domain updated:
kernel: [  324.383993] cfg80211:  DFS Master region: unset   
kernel: [  324.383994] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
kernel: [  324.383995] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A) 
kernel: [  324.383997] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
kernel: [  324.383998] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
kernel: [  324.383999] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
kernel: [  324.384000] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
kernel: [  324.384001] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
kernel: [  324.384002] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
kernel: [  324.384004] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
kernel: [  324.443676] usb 3-2: ath9k_htc: USB layer deinitialized
kernel: [  324.681297] usb 3-2: new high-speed USB device number 5 using xhci_hcd
kernel: [  328.948569] alx 0000:04:00.0: irq 52 for MSI/MSI-X
kernel: [  328.948899] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
kernel: [  329.878944] usb 3-2: New USB device found, idVendor=0cf3, idProduct=9271
kernel: [  329.878948] usb 3-2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
kernel: [  329.878950] usb 3-2: Product: UB91C
kernel: [  329.878951] usb 3-2: Manufacturer: ATHEROS 
kernel: [  329.878952] usb 3-2: SerialNumber: 12345
kernel: [  329.879408] usb 3-2: ath9k_htc: Firmware htc_9271.fw requested
kernel: [  329.879548] usb 3-2: firmware: direct-loading firmware htc_9271.fw
kernel: [  330.160798] usb 3-2: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
kernel: [  330.412471] ath9k_htc 3-2:1.0: ath9k_htc: HTC initialized with 33 credits
kernel: [  330.677966] ath9k_htc 3-2:1.0: ath9k_htc: FW Version: 1.4
kernel: [  330.677970] ath: EEPROM regdomain: 0x833a
kernel: [  330.677971] ath: EEPROM indicates we should expect a country code
kernel: [  330.677972] ath: doing EEPROM country->regdmn map search
kernel: [  330.677973] ath: country maps to regdmn code: 0x37
kernel: [  330.677974] ath: Country alpha2 being used: GB
kernel: [  330.677975] ath: Regpair used: 0x37
kernel: [  330.682429] ieee80211 phy2: Atheros AR9271 Rev:1
kernel: [  330.682438] cfg80211: Calling CRDA for country: GB
kernel: [  330.985324] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
kernel: [  330.985359] cfg80211: Regulatory domain changed to country: GB
kernel: [  330.985361] cfg80211:  DFS Master region: ETSI
kernel: [  330.985363] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
kernel: [  330.985365] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
kernel: [  330.985366] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
kernel: [  330.985368] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
kernel: [  330.985369] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
kernel: [  330.985371] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
##########################
Other info:
I have 3 usb on my MSI laptop, two 3.0 usb in left and one 2.0 usb in right, when i put the wifi card in the 3.0 usb port the disconections are much more often than when i use the right alone 2.0 usb port.
Im reading a lot of info about the problem and the direction is about power problem in  the usb port.
In windows all works fine. Im using the driver you have here latest version. I had another driver the 76c902360ef8a095133258fd0d287292 htc_9271.fw and the problem was the same.
TY :)

olerem, 2015-08-31:
Using new firmware with old driver/kernel will not do any changes. Please update kernel.

user113, 2015-08-31:
Advice: English no main language
Hi, i upgraded the kernel to debian backport 4.1.0-0.bpo.1-amd64 and im right now 3 hours without any disconnection in the usb.  Seems that works :D.

olerem, 2015-08-31:
If if there some disconnections, i hope it will restore within 10 seconds.

user113, 2015-08-31:
Advice: English no main language
After more test (all day) i think i can say 100% that with kernel 4.1.0-0.bpo.1-amd64 there is 0 disconnections. I dont know what changes have that kernel but is fixed.
Bye.

olerem, 2015-09-01:
glad to hear it :)

wice22, 2017-03-11:
Not sure if u are following this but I have similar issue, but with latest kernel + firmware
using: ALPHA **AWUS036NHA**
**dbud:**
```
[ 2076.850658] ath: phy454: Failed to wakeup in 500us
[ 2077.022912] usb 3-1: ath9k_htc: USB layer deinitialized
[ 2077.458764] usb 3-1: new high-speed USB device number 89 using ehci-pci
[ 2077.806076] usb 3-1: New USB device found, idVendor=0cf3, idProduct=9271
[ 2077.806081] usb 3-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 2077.806083] usb 3-1: Product: UB91C
[ 2077.806085] usb 3-1: Manufacturer: ATHEROS
[ 2077.806086] usb 3-1: SerialNumber: 12345
[ 2077.811670] usb 3-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 2077.812151] usb 3-1: firmware: direct-loading firmware ath9k_htc/htc_9271-1.4.0.fw
[ 2078.307289] usb 3-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[ 2078.576630] ath9k_htc 3-1:1.0: ath9k_htc: HTC initialized with 33 credits
[ 2079.379030] ath9k_htc 3-1:1.0: ath9k_htc: FW Version: 1.4
[ 2079.379035] ath9k_htc 3-1:1.0: FW RMW support: On
[ 2079.379037] ath: EEPROM regdomain: 0x833a
[ 2079.379039] ath: EEPROM indicates we should expect a country code
[ 2079.379041] ath: doing EEPROM country->regdmn map search
[ 2079.379042] ath: country maps to regdmn code: 0x37
[ 2079.379044] ath: Country alpha2 being used: GB
[ 2079.379045] ath: Regpair used: 0x37
[ 2079.397813] ieee80211 phy455: Atheros AR9271 Rev:1
[ 2079.423011] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2080.325742] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2080.371626] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2081.341801] usb 3-1: USB disconnect, device number 89
[ 2083.284344] ath: phy455: Failed to wakeup in 500us
```

It try's to reconnect again and again, 

**uname -r**
`4.6.0-kali1-amd64`

**apt-cache show firmware-atheros**
```
Package: firmware-atheros
Source: firmware-nonfree
Version: 20161130-2
Installed-Size: 8047
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Architecture: all
Suggests: initramfs-tools
Size: 3131328
SHA256: b1bb504275339f6218eb307bcddfb4f2bdb709d18ce9cc2333d04bc3b0a00875
SHA1: 54e11f24724bd76e33d5906f09698c4de72f9cb2
MD5sum: 54df1d1d7530cbe1cc1b134c6db484a0
Description: Binary firmware for Atheros wireless cards
 This package contains the binary firmware for USB wireless network and
 Bluetooth cards supported by the ar5523, ath3k, ath6kl_sdio, ath6kl_usb,
 ath9k_htc or ath10k drivers.
 .
 Contents:
.................................
  * ath6k/AR6004/hw1.3/fw-3.bin
  * Atheros AR7010 firmware, version 1.4.0 (ath9k_htc/htc_7010-1.4.0.fw)
  * Atheros AR9271 firmware, version 1.4.0 (ath9k_htc/htc_9271-1.4.0.fw)
  * Atheros AR7010 firmware, version 1.3.1 (htc_7010.fw)
  * Atheros AR9271 firmware, version 1.3.1 (htc_9271.fw)
  * Qualcomm Atheros QCA61x4 version 3.0 UART BT NVM configuration
    (qca/nvm_00130300.bin)
...............................
Description-md5: cc64d612094e73015861cf2915345869
Multi-Arch: foreign
Homepage: https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git
Tag: admin::hardware, role::app-data, use::driver
Section: non-free/kernel
Priority: optional
Filename: pool/non-free/f/firmware-nonfree/firmware-atheros_20161130-2_all.deb
```

olerem, 2017-03-11:
@wice22, please read: https://github.com/qca/open-ath9k-htc-firmware/wiki/usb-related-issues

wice22, 2017-03-11:
hmm.. I'll try all of them now :)

wice22, 2017-03-11:
also wondering about the firmware, u see system loads old and new firmware at the same time,  version 1.3.1  and 1.4.0, may be it's creating some sort of issues too ? some conflict?

cheers

psyborg55, 2019-01-13:
i've had this problem too with alfa card, especially if enabled more channels e.g. 2.3-2.7GHz. eventually it's reception sensitivity went bad showing 12-15dB less signal. after replacing faulty T/R switch it now works without problems with even wider range set (2192-2992) and i noticed no more disconnects. so, most likely it is hardware fault

erikarn, 2019-01-13:
Yeah, the hardware isn't designed for 2.3-2.7, so a bunch of weird crap can
go wrong. Sorry :(

-a

psyborg55, 2019-01-18:
no such crap with tl-wn722

TheMrRandomDude, 2019-09-22:
@psyborg55 Hey psyborg. I'm also having an issue with my Alfa AWUS036NHA. I tried to get it working on Debian and Windows. I also went as far to booting a live Kali usb and trying it there and it doesn't work.

What happens is that the led doesn't turn on at all, then it does for one second and the device shows up in ifconfig and on windows network adapters, and then the led turns off and stays off for 3-15 seconds (random) and this goes in a loop.

I'm going crazy. I checked all components with a multimter except for big chips like Atheros chip, and the EEPROM chip and other small ones I didn't know how to test. I didn't even know that I have a T/R switch on it.

Could I please ask you to lend me an advice or two on fixing it? Thank you :)

psyborg55, 2019-09-22:
first, try using cable other than the one you got in package (suggestion 50cm or shorter)
if it still doesn't work, enable debug on linux system and post output here

TheMrRandomDude, 2019-09-22:
@psyborg55 Tried multiple cables too, didn't work. I thought that it might require more power and ask for a powered hub, but I saw people using it on Youtube with a regular old 2.0 USB. It should work normally.

By debug I suppose you meant my dmesg output? I hope I got it right. Thank you so much once again

**dmesg**

https://pastebin.com/T0Cg6SpW

psyborg55, 2019-09-23:
try to get full debug output, by passing debug parameter to driver. e.g. on my system it is in /etc/modprobe.d/ath9k.conf
options ath9k_htc debug=0xffffffff

Hackndo, 2020-07-08:
@psyborg55 Thank you for suggesting a cable change. I was trying to debug @TheMrRandomDude exact same issue with my card, and other cards, on multiple OS, virtualbox, live USB and so on.
Then I read your answer, tried a new cable, and now it works like a charm.

Apteryks, 2023-11-30:
8 years later, I'm seeing this error on a Lenovo X200 laptop.  It renders my ThinkPenguin TP150N dongle unusable:

```
[ 1862.993642] usb 5-2: ath9k_htc: Firmware htc_9271.fw requested
[ 1863.279534] usb 5-2: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
[ 1863.530405] ath9k_htc 5-2:1.0: ath9k_htc: HTC initialized with 33 credits
[ 1863.800067] ath9k_htc 5-2:1.0: ath9k_htc: FW Version: 1.4
[ 1863.800076] ath9k_htc 5-2:1.0: FW RMW support: On
[ 1863.800079] ath: EEPROM regdomain: 0x65
[ 1863.800081] ath: EEPROM indicates we should expect a direct regpair map
[ 1863.800084] ath: Country alpha2 being used: 00
[ 1863.800085] ath: Regpair used: 0x65
[ 1863.804853] ath: EEPROM regdomain: 0x807c
[ 1863.804860] ath: EEPROM indicates we should expect a country code
[ 1863.804862] ath: doing EEPROM country->regdmn map search
[ 1863.804863] ath: country maps to regdmn code: 0x3a
[ 1863.804865] ath: Country alpha2 being used: CA
[ 1863.804867] ath: Regpair used: 0x3a
[ 1863.804868] ath: regdomain 0x807c dynamically updated by country element
[ 1863.805298] ieee80211 phy3: Atheros AR9271 Rev:1
[ 1863.833030] ath9k_htc 5-2:1.0 wlp0s29f7u2: renamed from wlan0
[ 1870.071792] usb 5-2: USB disconnect, device number 11
[ 1870.133299] ath: phy3: Failed to wakeup in 500us
[ 1870.143484] ath: phy3: Failed to wakeup in 500us
[ 1870.199311] usb 5-2: ath9k_htc: USB layer deinitialized
[ 1870.471414] usb 5-2: new high-speed USB device number 12 using ehci-pci
[ 1870.819202] usb 5-2: device descriptor read/64, error -71
[ 1872.595141] usb 5-2: device descriptor read/64, error -71
[ 1873.967477] usb 5-2: new high-speed USB device number 13 using ehci-pci
[ 1874.811473] usb 5-2: device descriptor read/64, error -71
[ 1875.275477] usb 5-2: device descriptor read/64, error -71
[ 1875.383538] usb usb5-port2: attempt power cycle
[ 1875.875438] usb 5-2: new high-speed USB device number 14 using ehci-pci
[ 1876.303477] usb 5-2: device not accepting address 14, error -71
[ 1877.143475] usb 5-2: new high-speed USB device number 15 using ehci-pci
[ 1877.571406] usb 5-2: device not accepting address 15, error -71
[ 1877.571497] usb usb5-port2: unable to enumerate USB device
[ 1877.835381] usb 6-2: new full-speed USB device number 3 using uhci_hcd
[ 1878.021406] usb 6-2: not running at top speed; connect to a high speed hub
[ 1878.052199] usb 6-2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[ 1878.052221] usb 6-2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 1878.052228] usb 6-2: Product: UB93
[ 1878.052235] usb 6-2: Manufacturer: ATHEROS
[ 1878.052241] usb 6-2: SerialNumber: 12345
[ 1878.056376] usb 6-2: ath9k_htc: Device endpoint numbers are not the expected ones
```

Since the e1000e wired ethernet also appears to be flaky on that X200, I don't have connectivitity at the moment, so I wasn't able to upgrade the OS to see if it would help things.  I'll try later using USB tethering.

Apteryks, 2023-12-01:
@psyborg55 I tried it from another Guix System laptop, and I had a different USB error but it also would not work, so your theory of a HW issue seems plausible. It's the first time I experienced such failure from this kind of electronic though, so I'm surprised.  My adapter uses the AR9271 chipset (see: https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb).

psyborg55, 2023-12-31:
i restored my awus036nha fully. no more any of these messages with old laptop and two routers. parts that needed replacement (except for richtek power converter) are VLS252012ET-2R2M and only one was faulty (1.2V i guess) according to LCR measurement. also tried to use parts from mq4wu5108 card but these did not work, i think they are of different value than 2.2uH

Apteryks, 2023-12-31:
@psyborg55 just to be clear, you needed to replace 1. richtek power converter integrated circuit (which model?) 2. one of the two VLS252012ET-2R2M, right?

For other readers, the VLS252012Et-2R2M part is a 2.2 uH surface mounted inductor (see: https://product.tdk.com/system/files/dam/doc/product/inductor/inductor/smd/catalog/inductor_automotive_power_vls252012e-ca_en.pdf)

psyborg55, 2023-12-31:
yes, those were busted completely. it is RT8020GQW.
AR9271 got damaged too (about 0.7w higher power consumption) but continued to work so i replaced it as well (put it on my tl-wn722 that i used entire time)
i also replaced SE2576 power amp, but it appears that one wasn't damaged

Apteryks, 2024-01-03:
I just opened my tpe-n150usb adapter from ThinkPenguin; it was a bit difficult to open: I used a small flat screwdriver to push the black plastic pin on the metal sleeve then pried around to push back the black plastic casing.

Here are some shots of the board; I trust the bigger '13 NCJI C3101075X01' identified chip is the AR9271, and I also identified an STMicroelectronics 24C08RP EEPROM (8/16KB) and a BDC 9706A DC/DC voltage converter (https://www.diodes.com/assets/Datasheets/products_inactive_data/AUR9706.pdf).

[image]
[image]
[image]

Perhaps I should probe with a DMM around the converter to verify the voltages are expected.

Apteryks, 2024-01-03:
Hm, I've tried connecting it, but it's difficult while its case is disassembled: I shimmed it into the USB port using some thin polystyrene foam sheet.  It appears completely dead (dmesg is silent), whether I connect it directly (the usual case) or via a 1 m USB extension cord.  I hope I haven't made it worse :-).

I'm lacking lab instruments here to do the measurements, unless I can somehow use my computer as both a function generator and an oscilloscope to use one of the tricks explained here: https://www.wikihow.com/Measure-Inductance (I think I've done so in the past for the scope part, using xoscope). audmes looks like another option for the oscilloscope part.

Apteryks, 2024-01-29:
@psyborg55 I'm not well equipped/motivated enough to fix mine.  If you are interested, I could ship this to you free of charge; otherwise it'll soon find its way to the dust bin.