qca/open-ath9k-htc-firmware #73: no option for forcing ar9271 in 802.11g only mode

yipperr, 2015-02-15:
there is no option to disbale N mode and force 802.11g connection mode on an alpha awus036nha wireless adapter

modinfo ath9k_htc only list these four parameters

parm:           debug:Debugging mask (uint)
parm:           nohwcrypt:Disable hardware encryption (int)
parm:           btcoex_enable:Enable wifi-BT coexistence (int)
parm:           ps_enable:Enable WLAN PowerSave (int)

no parameter to disable N mode in hardware

olerem, 2015-02-16:
It is useless option. Search for "set bitrates" commands on this page:
https://wireless.wiki.kernel.org/en/users/documentation/iw

Other question is, if set bitrate is actually working with this driver or firmware. If not, any one is welcome to provide it.

yipperr, 2015-02-16:
sorry for not mentioning this set bit-rates command does not work its values are ignored completely both legacy-2.4 and mcs 2.4 indexes are never stick  also i have noticed that the driver constantly changes the bitrates while in windows the driver is never changed from the 65mbps bitrate 

also the linux driver always defaults to 40mhz N mode while the windows always stays in 20mhz in N mode and there is no way to force 20mhz mode or force 802.11g with iw or iwconfig as those api are not considered by the driver

yipperr, 2015-02-16:
any ideas ?

olerem, 2015-02-16:
I need work, all workers are welcome. So if in need to be done, why not implement it correctly from beginning? :)

yipperr, 2015-02-16:
i was afraid of that i am just learning c++ and  reading on linux kernel interfaces and driver so it will take time and lot of it also would be cool to come up with static rate settting also the auto one is annoying 

thanks for your help

olerem, 2015-02-16:
If you need help, contact me :D

ballo, 2017-06-30:
Hey, it seems that set bitrates is an ignored option with this driver. Can someone please confirm so I know to stop trying to get Linux to work and try a different OS?

olerem, 2017-06-30:
bitrate filter can be set only with latest git version of this firmware. You should compile it by your self. No binary is currently available.

ballo, 2017-06-30:
Oh, OK. You mean open-ath9k-htc-firmware-1.4.0?

olerem, 2017-06-30:
yea, i mean version version never then 1.4.0

ballo, 2017-07-01:
OK. Pardon my ignorance but I can reuse my /toolchain/ from 1.4? Having to re-download and re-compile that whole thing would be a pain

ballo, 2017-07-01:
Using the firmware compiled from github:
[78964.599143] usb 1-1: new high-speed USB device number 13 using ehci_hcd
[78965.066970] usb 1-1: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
[78965.325095] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[78965.718849] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[78965.718858] ath9k_htc 1-1:1.0: ath9k_htc: Please upgrade to FW version 1.3
[78965.722530] Failed to initialize the device
[78965.734587] ath9k_htc: probe of 1-1:1.0 failed with error -22

ballo, 2017-07-01:
I used the ugly hack on post #2 here to get it to load:
https://github.com/qca/open-ath9k-htc-firmware/issues/7

however, it still does not bit rate limit using either iwconfig wlan0 rate or iw dev wlan0 set bitrates

olerem, 2017-07-01:
So you are using old kernel. What are you expect to happen?

ballo, 2017-07-01:
It's difficult to upgrade the kernel without upgrading the whole distro which means systemd which I refuse to touch with a 20ft pole. I got around this last time by compiling the driver module separately but that doesn't appear to be an option here.

paulfertser, 2017-07-01:
ballo, what exactly is incompatible in newer kernel with older distro parts? I think you should be able to use it just fine. Regarding systemd, are you sure your distro of choice doesn't provide an alternative? And in case you really want to keep the old kernel, you can use kernel backports to get latest drivers working with it.

ballo, 2017-07-01:
I'll have to revisit this issue when I have 10,000 as much bandwidth as I currently do