qca/open-ath9k-htc-firmware #75: (question) Power saving mode in Ad Hoc

Fernuts92, 2015-04-24:
What kind of Power saving mode this driver and its firmware have implemented for ad hoc networks?

I would like to be able to change some features of the Power Saving control, specifically, I wanted to be able to put the nodes sleeping or waking up at will.

Is this possible using this driver and firmware?

olerem, 2015-04-25:
It depends mostly on USB Host Cotrnoller if it will be able to completely power off/suspend the device. If host can't do it, then adapter will go to firmware pre-load state. It will consume less power...
No other power save functionality is enabled.

Fernuts92, 2015-04-25:
I understand but how could I instruct the adapter to enter in sleep state (remaining associated to the ad hoc cell) and to wake up from that sleep state? I'm doing my own Power saving protocol so I really need to be able to control this transitions. However, the driver/firmware does not accept any commands related to power management so I have no clue on how to accomplish this kind of control.. 

Do you know of any file or functions,  in the firmware or in the ath9k_htc driver source code, that, if changed, could enable this?

Thank you in advance.

olerem, 2015-04-25:
Please, be more specific. Describe your own Power saving protocol.

Fernuts92, 2015-04-25:
Basically, I use an out-of-band control channel to tell a node that it should transmit. My goal is to turn the nodes into sleep state while they are idle, waiting for the order to start transmitting that comes from the out of band channel. I could turn off the adapter but then it would take too much time to reconnect back to my ad hoc network. Therefore, I need a way of switching the adapter into sleep state and quickly switch it back to active state (in short, put the adapter sleeping while associated to the ad hoc cell). 

Firstly, I was trying to achieve this with iwconfig commands that allow the configuration of sleep periods and transmitting periods but  those commands are not accepted.

Now I was wondering if there is any solution, using the existing driver and firmware, to be able to fulfill my requirements. I suspect that there is nothing already done but I think that is possible to adapt the existing code to my needs, maybe adapting the power saving mode implementation used in infrastructure mode. However, I do not have any experience with drivers so things are a little complicated right now.

olerem, 2015-04-26:
First of all, please forget about iwconfig. Use iw.
So, you are talking about Dynamic Power Save, load ath9k_htc module with ps_enable=1 and i wish happy bugs hunting ;)
Power consumption of ath9k_htc devices is a sum of microcontroller CPU + WiFi MAC + WiFi PHY. If i see it correctly ps_enable will affect only WiFi PHY or WiFi MAC. So don't expect match power saving.

afilini, 2016-02-29:
I'm kinda interested in this too, I need to create a p2p group while being connected to another access point. Using a PCI ath9k antenna everything works fine (by loading the module with use_chanctx=1), with the ath9k_htc I got it to work only once by enabling power save mode. Are there any progress on this side? I am also interested in learning how this low level components works, if it's not too hard I can try fixing the code too, but I need some advices from all of you

olerem, 2016-02-29:
Just stop worrying and start doing ;)
Find you and adapter with accessible UART lines, bay an usb to uart dapter, do some soldering and explore it :D