qca/open-ath9k-htc-firmware #81: Is this really free software?

mrjpaxton, 2016-01-03:
Sorry, I don't mean to intrude with my question, but I have to know. Is this really free software, or is it **just** open source? I question most of the source code that says it's from companies developing IP chips (like Xtensa). Is this proprietary code required to build the two firmwares?

It says they can't be modified, decompiled, etc... That doesn't make it a free software license.

I just want to know if I need to throw my ath9k_htc adapter in the trash and buy an actual free one, because Debian doesn't even have this firmware in their free repos.

One good place I like to check is Parabola. So I looked, and they have it under Libre:

https://www.parabola.nu/packages/libre/any/ath9k-htc-firmware/

And it's also dependent on these:

https://www.parabola.nu/packages/libre/i686/xtensa-unknown-elf-binutils/
https://www.parabola.nu/packages/libre/i686/xtensa-unknown-elf-gcc/

But again, that is just the community speaking. I don't care if a whole bunch of people call it free software, but I'm making a claim that this software is non-free based on the license agreements that I've been reading from Tensilica. I wonder what RMS would say about this firmware...

olerem, 2016-01-03:
I'm not sure how to classify your question:
- you don't wont to intrude but deciding to do so
- you have complete code of firmware but don't read LICENSE.txt
- even with complete code for some reasons wont to decompile it
- promising to harm your self if we tell you that is is open but not completely free software, at same time you are writing this question on hardware which has closed source non free components with potential back doors.
Sure you are free to throw this adapter in the trash because debian decided to put opensource firmware to same repo with not opensource FWs.

n1000, 2016-01-03:
I'm not sure I understood the original comments fully, but maybe these files are the ones in question:

https://github.com/qca/open-ath9k-htc-firmware/tree/146bff1075a9f62c5d64c984bf6db502f8cd2671/sboot/include/xtensa

and

https://github.com/qca/open-ath9k-htc-firmware/tree/146bff1075a9f62c5d64c984bf6db502f8cd2671/target_firmware/magpie_fw_dev/target/inc/xtensa

The license.txt mentions that these files are released under the MIT license, but the license header in the files is not the MIT license.

Maybe this is the source of the confusion.

erikarn, 2016-01-03:
That's because we missed some of the conversions when we imported it. We
have the OK for MIT'ing all of the tensilica stuff that's in this tree.

It's open. Sheesh. :)

-a

olerem, 2016-01-04:
I talked with some people, one point of initial message seems to be a real bug.
debian maintainers pack all atheros FWs (except of carl9170) to one package. For this there are fallowing reasons:
- initial ath9k_htc fw (firmware/htc_*) is not open source.
- opensource version is located in firmware/ath9k_htc/*

So, what should be done:
- firmware/WHENCE should be updated. Paul provided a patch for this. @erikarn, can you please take a look at it or forward it to fw git?
- bug report for debian maintainers. Paul promised to do this as well after first step is done.

paulfertser, 2016-01-12:
For the reference:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=711470
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810047
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810672

Legimet, 2016-01-30:
Another wrong license header can be found in target_firmware/wlan/ah_osdep.h, and is also in NOTICE.TXT. See #45.

olerem, 2016-11-20:
this issue can be closed now, see:
https://ftp-master.debian.org/new/open-ath9k-htc-firmware_1.4.0-81-gf206e56+dfsg-1.html

mimi89999, 2020-09-28:
Hello,
This firmware was included in Debian main repo and repos of FSDG distros. Most RYF certified WiFi adapters https://ryf.fsf.org/categories/wireless-adapters are using chips supported by this firmware with this firmware. I even don't know what other WiFi adapter firmware is considered free and what adapters are using it.
While it is possible that all the people who reviewed this firmware and those devices missed some important freedom issues, I think that this firmware can really be considered free software unless demonstrated otherwise.