qca/open-ath9k-htc-firmware #83: Disable ACKs/Retransmissions

Seitam, 2016-01-27:
For study purposes I would like to know if it is possible to disable the ACKs and retransmissions by modifying the ath9k_htc driver/firmware. Some academic publications explain how they did it in the past  by means of the MadWifi driver. Nowadays, with the MadWifi project deprecated I don't know if it is is possible to do it with the current replacement drivers (I am specially interested with the ath9k_htc for the usb support).
After an intensive research on the web I haven't found any evidences that this is doable. 

Any help would be greatly appreciated.

danzipie, 2016-02-12:
Hi, you could be interested in changing the HAL_TXDESC_NOACK flag on the transmitter side.

rodrigowue, 2021-01-07:
Hey @danzipie and @Seitam ,

Could you give me more info on how to enable the HAL_TXDESC_NOACK flag, please? 
I'm a newbie on this and I'm not sure how to do that.

I've seen that the HAL_TXDESC_NOACK starts with the value 2 by default.
-----------------------------------------------------------
File: open-ath9k-htc-firmware/target_firmware/wlan/ah_desc.h
-----------------------------------------------------------
Content:
...
#define HAL_TXDESC_NOACK        0x0002
...
-----------------------------------------------------------

To enable it, should I change it to another value?

Cheers,
Rodrigo