qca/open-ath9k-htc-firmware #90: make sure this firmware is supplied as free/opensource package on debian

olerem, 2016-08-20:


olerem, 2016-08-20:
This bug report was made back in 11 Jun 2014, last update at 23 Jan 2015.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=751339

the main problem is building a toolchain which is not trivial for packaging

olerem, 2016-08-20:
Link all related bug reports:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=711470
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810047
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810672

olerem, 2016-09-08:
Contacted gcc devs:
https://www.mail-archive.com/gcc@gcc.gnu.org/msg80938.html

patches for GCC could be accepted. If this would be done, then we don't need to create own patched toolchain.

olerem, 2016-09-16:
starting research on GCC and Bintuils. First of all, updated firmware to latest GCC version. If i will need to make gcc patches an push it upstream i need to be able to test it.

So, here is the pull request:
https://github.com/qca/open-ath9k-htc-firmware/pull/91

olerem, 2016-09-21:
Side note: we have no test automation :( https://github.com/qca/open-ath9k-htc-firmware/issues/95

olerem, 2016-09-21:
https://github.com/qca/open-ath9k-htc-firmware/pull/91 is done. Now we need continue to research binutils and GCC issue.

olerem, 2016-09-21:
According to irc #debian-devel it should be possible with less problems to provide package for this firmware. Keywords are: binutils-source and gcc-6-source. Here are reverse depnedencies for this packages which should be a good example to start:

[0] 11:22 cbe@benz:~ master $ build-rdeps binutils-source
WARNING: dose-extra >= 4.0 is not installed. Falling back to old unreliable behaviour.
Reverse Build-depends in contrib:
---------------------------------

No reverse build-depends found for binutils-source.

Reverse Build-depends in main:
------------------------------

cross-toolchain-base
binutils-z80
cross-binutils
cross-toolchain-base-ports
openbios
binutils-mingw-w64
binutils-arm-none-eabi

Found a total of 7 reverse build-depend(s) for binutils-source.

Reverse Build-depends in non-free:
----------------------------------

No reverse build-depends found for binutils-source.

[0] 11:39 cbe@benz:~ master $ build-rdeps gcc-6-source
WARNING: dose-extra >= 4.0 is not installed. Falling back to old unreliable behaviour.
Reverse Build-depends in contrib:
---------------------------------

No reverse build-depends found for gcc-6-source.

Reverse Build-depends in main:
------------------------------

gcc-6-cross-ports
gnat-mingw-w64
gcc-6-cross
gcc-mingw-w64

Found a total of 4 reverse build-depend(s) for gcc-6-source.

Reverse Build-depends in non-free:
----------------------------------

No reverse build-depends found for gcc-6-source.

Legimet, 2016-10-04:
I hope this can be included in Debian stretch.

olerem, 2016-10-06:
next pre debian PR: https://github.com/qca/open-ath9k-htc-firmware/pull/104

olerem, 2016-10-06:
@Legimet , sure... it is not trivial and frustrating work which take lots of time. I doubt i'll do it again any time soon.

olerem, 2016-11-20:
This issue can be closed now, see:
https://ftp-master.debian.org/new/open-ath9k-htc-firmware_1.4.0-81-gf206e56+dfsg-1.html

closed 2016-11-20 by olerem