qca/open-ath9k-htc-firmware #107: Frequent connection drops.

pkgodara, 2016-10-18:
I am using TP-Link WN722N usb wireless adaptor on ubuntu 16.04( latest ).
Device 011: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n .

It is working good on windows but on ubuntu , connection drops frequently and I have to replug or restart network manager to make it work fine again.
I thing it doesn't update connection and also signal strength doesn't update on desktop.
Whatever is the problem , please try to fix is.

olerem, 2016-10-18:
I have ubuntu 16.04 and use this adapter for long distance connections, in places where my build in adapter is not able to get good signal. It works perfectly... so What i'm doing wrong? Wrong encryption, too new kernel? Too recent firmware?

Is your linux installed directly on HW or in VirtualBox?

pkgodara, 2016-10-19:
Its dual booted with windows 8.1 , i.e. directly installed on HW. I am using it on HP 15-ac053tx notebook.

olerem, 2016-10-19:
Please, try fallowing, take this file https://github.com/olerem/ath9k-htc-firmware-blob/blob/master/htc_9271.fw and put it instead of /lib/firmware/ath9k_htc/htc_9271-1.4.0.fw on you notebook. Make a backup of original fw first.
then reload ath9k_htc driver or restart notebook.
If no changes, please try some recent kernel, for example this one:
http://kernel.ubuntu.com/~kernel-ppa/mainline/daily/current/

closed 2016-10-25 by pkgodara