qca/open-ath9k-htc-firmware #108: AR9271 power pins

doru91, 2016-10-21:
Hello, everyone

I want to measure the power consumption of the AR9271 chip mounted inside a TP-Link TLWN722N dongle. In order to avoid measuring the power consumption of the USB sub-layer, my plan is to disassemble the dongle, access the 68 internal pins of the chip then connect the grabbers of my power monitor device to the power pins (VDDP33/GND - page 12 from datasheet) of the wifi chip:

- The datasheet states that pins 17, 26, 33, 47 represents digital 3.3 V power supply - VDDP33. My power monitor device has only one grabber for VCC. Which pin should I use? What's the reason for having multiple VDDP33 pins?

- Is there any better solution for measuring the power consumption of the chip?

- How did you obtain the power consumption parameters illustrated in page 144 of the datasheet?

Thank you,
Doru

doru91, 2016-10-24:
Is this the correct place for asking this question? :)

doru91, 2016-10-31:
Is this project dead?

I also asked a similar question on the ath9k_htc_fw mailing list but no moderator bothered to approve the question.

At this moment I am interested in the answer for the third bullet.

olerem, 2016-10-31:
I would say, you won't find here any one who created this documentation or some how participated in electrical part of design. If no one is answering, then no one can provide you an usable answer.

rfelten, 2016-12-05:
Hi Doru,
what do you mean by "power consumption of the USB sub-layer"?

Instead of de-solder four VDDO33 pins, you maybe want to intercept the 3.3V rail after the step down converter? It should be the trace right of "4R7", see https://wikidevi.com/w/images/8/8e/TL-WN722N-front_b.jpg.

> What's the reason for having multiple VDDP33 pins?

This allows the PCB designer to create a better PCB (in terms of RF immunity / RF emissions / signal crosstalk etc). Maybe "appnote pcb multilayer rf design" on Google will give some insights.

Please share your findings.

Image from https://wikidevi.com/wiki/TP-LINK_TL-WN722N

doru91, 2017-02-18:
Hi, Robert

Sorry for the extra-delayed answer :)

By "power consumption of the USB sub-layer" I meant that I wanted to measure the power consumption of the WiFi chip only. That's because I wanted to see clearly the power level corresponding to 'NETWORK_SLEEP' and 'AWAKE' states [1] and I thought that the hardware components around the WiFi chip will introduce noise and I won't see clearly these level. However, I've been able to do my measurements using the power consumption of the entire dongle. 

See [2] and [3] for a wiki with my findings.

If I have time, I'll also try your suggested solution for power measurement.

[1] http://lxr.free-electrons.com/source/drivers/net/wireless/ath/ath9k/hw.h?v=3.13#L438
[2] https://github.com/doru91/linux-stable/wiki/Single-Interface-Power-Save
[3] https://github.com/doru91/linux-stable/wiki/Two-Interfaces-Power-Save

rfelten, 2017-02-21:
Thank you for sharing the results. And good luck with your thesis!