qca/open-ath9k-htc-firmware #109: advanced tuning options support

psyborg55, 2016-11-01:
Hi

I've tried setting one of the advanced tuning options under wifi-iface section but it did not do anything.

The option bursting '1' had no effect on throughput. I suspect other similar options are also not supported yet such as compression, fast frames, xr etc.

Is there any plan to add basic support for these features or is there some patchwork for ath9k that could be our starting point? I've read on openwrt forums some users reported these features worked with ath9k.

olerem, 2016-11-02:
Hi, No there are no plans to do this. Patches are welcome ;)

psyborg55, 2016-11-02:
And why not? Seems like there is some code but related to 802.11n mode. What about legacy mode?

olerem, 2016-11-02:
No time.

erikarn, 2016-11-02:
hiya,

bursting should be easy to do - it's just WME configuration.

Fast-frames shouldn't be hard - but it needs ath9k/mac80211.

There's no compression support in 11n chips, sorry.

There's no official XR rates support in 11n chips, sorry

-a

psyborg55, 2016-11-02:
Hi

thanks for information

compression is not really important since it does not help always, as i've read it can even decrease performance.

XR from what i've found was never really used, for clarification would be nice to know what it really translates to, eXtended range or as somebody claims eXtended Receive, +20dB receiver sensitivity improvement

i've tried enabling fast frames on a madwifi device, no improvement in throughput when using ath9k_htc card. so i guess it's not of a big importance.

but enabling burst on the same device and using same ath9k_htc card managed to get increased throughput from ~20Mbps to ~23.5Mbps, of course in one direction only.

does this require simple code change like in the above example with ath6kl driver or there is more code required to support this?

erikarn, 2016-11-02:
The mac80211 layer needs to assemble fast frames and hand them frame at a
time down to the firmware. The firmware deals with 802.11 frames.

-adrian

psyborg55, 2016-11-15:
another thing found not to work is channel bandwidth 5/10 MHz. this htc driver is a complete joke..

olerem, 2016-11-15:
sore.. here is source, fix it.

psyborg55, 2018-08-06:
bursting works,
ani works,
dynack works,
5/10 channel width added ->https://github.com/psyborg55/linux/commit/9c9820bce1bf773cc9f6c403a8cc7fe53b1b4884

closing

closed 2018-08-06 by psyborg55