qca/open-ath9k-htc-firmware #112: AR9271 ping spikes

linux-wireless-misery, 2016-11-08:
I am experiencing disruptive ping spikes every 15-20 seconds lasting 2-3 sec. Over 500ms or outright packet drops. Have the same behaviour with 2 separate AR9271 USB cards (Netgear WNA1100 and TP-Link TL-WN722N).

Tested on Lubuntu 16.x, Fedora 24, Kali 2016.1 and 2. Kernels 4.1 to 4.8. Two separate computers.
It also happens on Windows, but:
-the spikes happen in similar timeframes, but aren't nearly as disruptive (two or three ~200ms roundtrips, rarely any packet loss)
-it gets fixed completely with Streaming Mode enabled in WLAN Optimizer (http://www.martin-majowski.de/faq.html)

Any ideas as to what's wrong?

olerem, 2016-11-08:
1. channel scan.
Depending on your usb host controller channel scan can take from 8 till 30 seconds. The interval of scan is configured by wpa_suplicant.
2. USB 3 controller performs better then USB2, even if this is USB2 adapter.
3. led blinking.
disable it and you will get better results.

olerem, 2016-11-08:
4. don't forget to update firmware. latest firmware can provide some optimisation if bound with latest kernel.

satvilla, 2016-11-08:
I have the same problem.
I try to compile the firmware with no luck; I also try to disable hwcrypt.

How can I disable the led blinking or change the wpa_suplicant?

Thanks in advance.

olerem, 2016-11-09:
Then most probably it in an issue with the router firmware. Google for "WMM bad ping" or some thing like this.
In my tests i use  TL-WDR4300 with openwrt, and noticeable ping change is only each 3 minutes for max 10 seconds. The latency is higer, but no packet drops..

linux-wireless-misery, 2016-11-09:
@olerem, you were right at the outset, it is wpa_supplicant triggering CTRL-EVENT-SCAN. Now, the problem is that I can't fix it: wpa_cli doesn't respond to **set** or **reconfigure** commands out of the box, and I can't make wpa_supplicant.service read wpa_supplicant.conf. I even managed to delete /sbin/wpa_supplicant at one point somehow! What a f*****g mess.

linux-wireless-misery, 2016-11-09:
@satvilla I still get spikes after turning WMM off, but they're not as severe now (no packet loss anymore).

olerem, 2016-11-09:
Try to disable HT40 if enabled.

linux-wireless-misery, 2016-11-09:
No difference.

erikarn, 2016-11-09:
hiya,

it may just be network manager triggering background scans.

-adrian

satvilla, 2016-11-09:
@olerem I'm sure the problem is not the router, because it only happen with this wifi client.
My real problem is when I try to watch a video on HD over the wifi lan, each few minutes the video stops, because the buffer is empty, until the buffer grow to the 100%. It also happen with SD content but less often because the buffer is bigger.
The cause of that problem is the ping of the computer, I monitorize that and it change from 6ms to 600-1000+ ms every few minutes.

I want to try to disable the blink or the wpa_supplicant, if it can help, but I'm not an experienced linux user (in fact I'm proud to have been able to compile this firmware), how can I do that?

soyersoyer, 2017-09-21:
From: https://blogs.gnome.org/dcbw/2016/05/16/networkmanager-and-wifi-scans/

You can "lock your WiFi connection profile to the BSSID of your access point" by entering your access point's BSSID under "Edit Connections..."

It works, no more spikes.