qca/open-ath9k-htc-firmware #114: All kinds of weird problems when using more than one or two Atheros cards

rodizio1, 2016-11-20:
I've beeen trying to narrow this issue down for months now, but it's so weird, really don't know what to do further. Already tried disabling LED blinking, but that didn't seem to help much.

Sometimes cards don't get initialized, sometimes they do, but don't receive packets, sometimes they do and receive packets only for a second or two then they crash.

```
[   62.491439] device ec086b1c7834 left promiscuous mode
[   62.495125] ath: phy3: Set HW RX filter: 0x37
[   62.506186] device ec086b1c4645 left promiscuous mode
[   62.514370] ath: phy2: Set HW RX filter: 0x37
[   62.521136] device c46e1f21bbf1 left promiscuous mode
[   62.535178] ath: phy1: Set HW RX filter: 0x37
[   63.355188] device c46e1f21bbf1 entered promiscuous mode
[   63.368003] ath: phy1: Set HW RX filter: 0x37
[   63.375141] device ec086b1c4645 entered promiscuous mode
[   63.389368] device ec086b1c7834 entered promiscuous mode
[   63.392006] ath: phy3: Set HW RX filter: 0x37
[   63.392372] ath: phy2: Set HW RX filter: 0x37
[   66.795152] cfg80211: Verifying active interfaces after reg change
[   86.338279] device ec086b1c7834 left promiscuous mode
[   86.344201] ath: phy3: Set HW RX filter: 0x37
[   86.358185] device ec086b1c4645 left promiscuous mode
[   86.370162] device c46e1f21bbf1 left promiscuous mode
[   86.377554] ath: phy2: Set HW RX filter: 0x37
[   86.378298] ath: phy1: Set HW RX filter: 0x37
[   87.191206] device c46e1f21bbf1 entered promiscuous mode
[   87.210421] ath: phy1: Set HW RX filter: 0x37
[   87.215230] device ec086b1c4645 entered promiscuous mode
[   87.218165] ath: phy2: Set HW RX filter: 0x37
[   87.233157] device ec086b1c7834 entered promiscuous mode
[   87.237414] ath: phy3: Set HW RX filter: 0x37
[   94.811654] device ec086b1c7834 left promiscuous mode
[   94.829155] device ec086b1c4645 left promiscuous mode
[   94.841140] device c46e1f21bbf1 left promiscuous mode
[   94.911149] ath: phy3: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   94.911174] ath: phy3: REGISTER READ FAILED: (0x803c, -110)
[   94.929142] ath: phy2: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   94.929152] ath: phy2: REGISTER READ FAILED: (0x803c, -110)
[   94.941136] ath: phy1: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   94.941145] ath: phy1: REGISTER READ FAILED: (0x803c, -110)
[   95.011135] ath: phy3: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.011146] ath: phy3: REGISTER READ FAILED: (0x810c, -110)
[   95.029191] ath: phy2: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.029201] ath: phy2: REGISTER READ FAILED: (0x810c, -110)
[   95.041133] ath: phy1: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.041141] ath: phy1: REGISTER READ FAILED: (0x810c, -110)
[   95.111132] ath: phy3: Timeout waiting for WMI command: WMI_REG_RMW_CMDID
[   95.111152] ath: phy3: REGISTER RMW FAILED:(0x0034, -110)
[   95.129215] ath: phy2: Timeout waiting for WMI command: WMI_REG_RMW_CMDID
[   95.129224] ath: phy2: REGISTER RMW FAILED:(0x0034, -110)
[   95.141139] ath: phy1: Timeout waiting for WMI command: WMI_REG_RMW_CMDID
[   95.141148] ath: phy1: REGISTER RMW FAILED:(0x0034, -110)
[   95.211319] ath: phy3: Timeout waiting for WMI command: WMI_REG_WRITE_CMDID
[   95.211339] ath: phy3: REGISTER WRITE FAILED, multi len: 2
[   95.211351] ath: phy3: Set HW RX filter: 0x2137
[   95.229146] ath: phy2: Timeout waiting for WMI command: WMI_REG_WRITE_CMDID
[   95.229163] ath: phy2: REGISTER WRITE FAILED, multi len: 2
[   95.229172] ath: phy2: Set HW RX filter: 0x2137
[   95.241140] ath: phy1: Timeout waiting for WMI command: WMI_REG_WRITE_CMDID
[   95.241148] ath: phy1: REGISTER WRITE FAILED, multi len: 2
[   95.241157] ath: phy1: Set HW RX filter: 0x2137
[   95.699182] device c46e1f21bbf1 entered promiscuous mode
[   95.712196] device ec086b1c4645 entered promiscuous mode
[   95.728139] device ec086b1c7834 entered promiscuous mode
[   95.799152] ath: phy1: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.799175] ath: phy1: REGISTER READ FAILED: (0x803c, -110)
[   95.812125] ath: phy2: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.812137] ath: phy2: REGISTER READ FAILED: (0x803c, -110)
[   95.828149] ath: phy3: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   95.828164] ath: phy3: REGISTER READ FAILED: (0x803c, -110)
[   95.899155] ath: phy1: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[... 154 lines cut ...]

Next test again with two cards. This time I can succesfully bring them up and issue "ifconfig wlan0 promisc".

Now the weird thing is, when I set the cards to promiscous mode via ifconfig, it seems to _always_ trigger that REGISTER_READ_TIMEOUT message. However, sometimes the card is still able to receive packets, even though the message appeared. After running "ifconfig wlan0 promisc" a few times in a row though, it certainly stops working. 

I can also trigger this with only one card. I.e. start-up the system, configure card, run "ifconfig promisc" for maybe 5-10 times and the card crashes.

Some general infos:
- It seems to get worse with more cards: Two cards seem to work in 99.9% of cases, three cards in maybe 10%.

- Can be reproduced with TPLink 722N/Alfa AWUS036NHA (both AR9271) and TPLINK 822 V2 (AR9287)

- It seems related to other stuff using the USB bus. If the ethernet connection on the Pi is in use (which is also connected via USB internally), three cards almost never work.

- The same happens with other wifi cards (rt2800usb). 1x ath9k_htc plus 3x rt800usb works, 2x ath9k_htc plus 1x rt2800usb is flaky. 2x ath9k_htc plus 2x rt2800usb is even more flaky. 

- However, 4x rt2800usb without any Atheros cards is absolutely stable and works 100%.

- When plugging the cards slowly each after another (instead of having all three connected during boot-up), chances to get them working seem to be higher, but it doesn't always work.

- It seems to have gotten worse since I changed Kernel timer frequency to 1000Hz (in order to fix a different problem with packets being delayed on the USB-Bus or subsystem somewhere).

- It is not power-related or bad cables or something, I have even gone so far as soldering the adapters directly to the Raspberry USB Ports and powering the Pi and the cards directly from a 7.5 Ampere power supply connected to a large Lipo battery. Problem remains.

- The problems only occur during initialization of the cards or a few seconds after the startup of the rx program which puts the cards into promiscous mode. If the cards have "survived" past that, they run stable for hours or even days. But just killing and re-starting the rx program (or issuing "ifconfig wlan0 promisc") can make them crash again anytime.

olerem, 2016-11-20:
All REG related commands go over EP3 and EP4. This EPs was configured for some time as Bulk, then i added patches to revert this behavior back to Interrupt mode. Latest kernel and FW should include some optimization for this EPs.
Which kernel version and FW versions are you using, do this changes applied or not?
I can imagine different usb related issues:
- not enough power
- not enough usb bandwidth, especially if Bulk used instead of interrupt.
- bad usb signal, too long cable or bad connector

I assume, it should be possible to reduce this problems by mowing as match functionality as possible to the FW. So we avoid direct REG read/write operations.

Currently without FW changes, you can reduce EP3/EP4 load by disabling LED and bgscan, if this are enabled.

rodizio1, 2016-11-20:
Kernel is 4.4.11, ath9k_htc-firmware is version 1.4. 

LED blink is already disabled. It is definitely not a power or cable issue (see my initial post).

How do I disable bgscan? Couldn't find anything about it with a quick search. Or do you mean scans that get initiated by network-manager or wpa-supplicant etc.? 

Network-manager, wpa-supplicant and everything else that might mess with the cards is disabled. Only thing I do is bring up the interface, set monitor mode and channel (all via plain ifconfig/iwconfig) then I start the rx program that puts the cards into promiscous mode.

Regarding not enough USB bandwidth: Well, I have never really measured USB bandwidth on the Pi, but it works flawless with four Ralink cards. So far I didn't notice it being slow when copying files over the USB ethernet connection or to an USB memory stick.

The incoming data is something around 500 1024bytes packets per dongle, so not too much.

Regarding the packet size: I read somewhere, that USB max. packet size is 512bytes for bulk transfer. So assuming some overhead, I figured 1024 bytes ethernet packets will probably lead to 2x 512 bytes packets plus 1x 20bytes (or whatever low number) packets on the USB Bus, i.e. that what is called fragmentation in the IP world and is best to be avoided. But lowering the packet size to 900 didn't seem to make a difference.

olerem, 2016-11-20:
hm.. so, what do i need to reproduce this issue?

rodizio1, 2016-11-20:
I just downloaded the latest Raspbian lite image to have a fresh clean install. Could not reproduce the issue with putting the cards in promiscous mode when I gave it a quick try.

However, as soon as I plug in the 3rd AR9271 stick, my USB keyboard does strange things (like repeating characters that I did not press and being laggy) and stops working completely shortly after.

When starting the Pi3 with the three sticks already attached, the keyboard keeps working, but I end up with only one card being shown with iwconfig.

Not sure if that is something easy for you to reproduce. If I can do something to narrow it down some more I'll gladly do that.

rodizio1, 2016-11-21:
Did some more testing. One issue really seems to be switching the cards to promiscous mode.

I have now changed this line in the code of the rx program that receives the packets:
`interface->ppcap = pcap_open_live(name, 2048, **1**, -1, szErrbuf);`

to this:
`interface->ppcap = pcap_open_live(name, 2048, **0**, -1, szErrbuf);`

to disable putting the cards into promiscous mode, and the problem that the cards stop receiving traffic shortly after the rx program has been started sometimes is gone.

But the problem that cards don't get initialized if more than two or three cards are used is still there.
I'll compile a kernel with wireless and atheros debugging enabled on the fresh Raspbian image now and check further ...

olerem, 2016-11-21:
In promiscous mode the card will dump everything what it get from the air over usb. It means, depending on environment you will get different results.
Some project already filed becouse of this :)
Do you get same 2-3 card issue on simple Linux PC too?

rodizio1, 2016-11-21:
The rx program still works as intended with promisc mode disabled, I guess this is because wireless cards in monitor mode forward all traffic anyway (?) As far as I see it in the code, it sets a pcap filter and uses a special mac address to filter traffic:

`"radio[0x4a:4]==0x13223344 && radio[0x4e:2] == 0x55%.2x", port);`
Too much (other) traffic is not the problem, this also happens with antennas off or on channel 14.

I will try to see what happens on a linux notebook.

BTW (little offtopic): To optimize that further, would it in theory be possible to filter out traffic based on mac addresses already in the atheros firmware, so that it doesn't even reach the host? And basically don't listen to other packets at all, i.e. abort reception and listen again as soon as the header with macs has been read?

olerem, 2016-11-21:
s/promiscous/monitore you are right... where are my thoughts

IMO filtering in the firmware should be good point. @erikarn , what do you think? I think this should help some other users too.

rodizio1, 2016-11-21:
Compiled a kernel now with ath9k_htc debugging enabled on the fresh Raspbian lite image. Kernel has stock Raspbian config (just some un-needed unrelated drivers removed to speed up compilation on the Pi) and is the same version as the stock Raspbian one.

two sticks already plugged before power-on, then plugged the third stick:
```
[   71.512684] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
[   71.633826] usb 1-1.2: New USB device found, idVendor=0cf3, idProduct=9271
[   71.633846] usb 1-1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   71.633859] usb 1-1.2: Product: UB91C
[   71.633872] usb 1-1.2: Manufacturer: ATHEROS
[   71.633884] usb 1-1.2: SerialNumber: 12345
[   71.635278] usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   71.933879] usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   73.420618] ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[   73.598751] ath: phy2: serialize_regmode is 0
[   73.598888] ath: phy2: UNDEFINED -> AWAKE
[   73.610212] ath: phy2: Reading from EEPROM, not flash
[   74.712633] ath: phy2: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   74.712648] ath: phy2: Multiple REGISTER READ FAILED (count: 8)
[   75.712640] ath: phy2: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   75.712667] ath: phy2: Multiple REGISTER READ FAILED (count: 8)
[   77.671673] ath: phy2: Read Magic = 0xA55A
[   77.671693] ath: phy2: need_swap = False
[   77.671708] ath: phy2: Bad EEPROM checksum 0xd382 or revision 0x000e
[   77.671753] ath: phy2: Unable to initialize hardware; initialization status: -22
[   77.671764] ath: phy2: Unable to initialize hardware; initialization status: -22
[   77.671802] ath9k_htc: Failed to initialize the device
[   77.673514] usb 1-1.2: ath9k_htc: USB layer deinitialized
```

After some tries, three sticks connected before power-up worked, but when plugging the 4th one:
```
[   87.633090] usb 1-1.5.1: new high-speed USB device number 8 using dwc_otg
[   87.884476] usb 1-1.5.1: New USB device found, idVendor=0cf3, idProduct=9271
[   87.884499] usb 1-1.5.1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   87.884512] usb 1-1.5.1: Product: USB2.0 WLAN
[   87.884525] usb 1-1.5.1: Manufacturer: ATHEROS
[   87.884569] usb 1-1.5.1: SerialNumber: 12345
[   87.886007] usb 1-1.5.1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   88.186557] usb 1-1.5.1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   89.803089] usb 1-1.5.1: Service connection timeout for: 258
[   89.803104] ath9k_htc 1-1.5.1:1.0: ath9k_htc: Unable to initialize HTC services
[   89.803129] ath9k_htc: Failed to initialize the device
[   89.803970] usb 1-1.5.1: ath9k_htc: USB layer deinitialized
```

In both cases "modprobe -r ath9k_htc" made atleast the USB subsystem crash (ssh connection and usb keyboard didn't work anymore). I'll see that I configure the serialport to see what is going on then.

erikarn, 2016-11-21:
That's weird, seeing transfer timeouts is a big problem. Can you enable
some debugging or capture the USB traffic on that USB port?

-a

rodizio1, 2016-11-22:
The dmesg log above was with ath9k_htc module parameter debug=0xffffffff, can I enable more debugging somewhere?

Here is a .pcap made with the usbmon module while plugging in a third card (two were already connected during boot-up).
[usbcapture.zip](https://github.com/qca/open-ath9k-htc-firmware/files/607118/usbcapture.zip)

The corresponding dmesg output:
[   61.382340] usb 1-1.4: new high-speed USB device number 6 using dwc_otg
[   61.487370] usb 1-1.4: New USB device found, idVendor=0cf3, idProduct=9271
[   61.487398] usb 1-1.4: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   61.487407] usb 1-1.4: Product: USB2.0 WLAN
[   61.487414] usb 1-1.4: Manufacturer: ATHEROS
[   61.487422] usb 1-1.4: SerialNumber: 12345
[   62.489355] usb 1-1.4: ath9k_htc: Firmware ath9k_htc/htc_9271-1.18.0.fw requested
[   62.781667] usb 1-1.4: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.18.0.fw, size: 51008
[   66.283344] cfg80211: Verifying active interfaces after reg change
[   68.297143] ath9k_htc 1-1.4:1.0: ath9k_htc: HTC initialized with 33 credits
[   68.396340] ath: phy3: Timeout waiting for WMI command: WMI_REG_READ_CMDID
[   68.396349] ath: phy3: REGISTER READ FAILED: (0x4020, -110)
[   68.396357] ath: phy3: Mac Chip Rev 0x0f.3 is not supported by this driver
[   68.396364] ath: phy3: Unable to initialize hardware; initialization status: -95
[   68.396371] ath: phy3: Unable to initialize hardware; initialization status: -95
[   68.396401] ath9k_htc: Failed to initialize the device
[   68.397502] usb 1-1.4: ath9k_htc: USB layer deinitialized

Please note, this was _not_ made on the fresh Raspbian image, but on my EZ-Wifibroadcast image which has a different kernel with 1000Hz timer frequency (not sure if that makes a difference, I can do it again with the other clean Raspbian image with stock kernel if needed).

olerem, 2016-11-22:
can you recompile the kernel with this change?
diff --git a/drivers/net/wireless/ath/ath9k/hif_usb.h b/drivers/net/wireless/ath/ath9k/hif_usb.h
index 7c2ef7e..438701c 100644
--- a/drivers/net/wireless/ath/ath9k/hif_usb.h
+++ b/drivers/net/wireless/ath/ath9k/hif_usb.h
@@ -61,7 +61,7 @@ extern int htc_use_dev_fw;
 #define MAX_PKT_NUM_IN_TRANSFER 10
 
 #define MAX_REG_OUT_URB_NUM  1
-#define MAX_REG_IN_URB_NUM   64
+#define MAX_REG_IN_URB_NUM   8
 
 #define MAX_REG_IN_BUF_SIZE 64

rodizio1, 2016-11-22:
Tried the patch on the stock raspbian 4.4.32 kernel on the clean image now, I think it makes no difference.

This is a complete dmesg log from bootup with four adapters installed. Will do more testing ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.32-v7 (root@raspberrypi) (gcc version 4.9.2 (Raspbian 4.9.2-10) ) #1 SMP Mon Nov 21 21:53:32 UTC 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 807b31c0, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f62000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2709.boardrev=0x2a02082 bcm2709.serial=0xadfc03e7 smsc95xx.macaddr=B8:27:EB:FC:03:E7 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940188K/966656K available (5599K kernel code, 303K rwdata, 1544K rodata, 428K init, 749K bss, 18276K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80702008   (7145 kB)
      .init : 0x80703000 - 0x8076e000   ( 428 kB)
      .data : 0x8076e000 - 0x807b9cb0   ( 304 kB)
       .bss : 0x807bc000 - 0x80877628   ( 750 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000010] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000027] Switching to timer-based delay loop, resolution 52ns
[    0.000295] Console: colour dummy device 80x30
[    0.001340] console [tty1] enabled
[    0.001393] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001460] pid_max: default: 32768 minimum: 301
[    0.001793] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001836] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002791] Disabling cpuset control group subsystem
[    0.002847] Initializing cgroup subsys io
[    0.002898] Initializing cgroup subsys memory
[    0.002962] Initializing cgroup subsys devices
[    0.003004] Initializing cgroup subsys freezer
[    0.003045] Initializing cgroup subsys net_cls
[    0.003116] CPU: Testing write buffer coherency: ok
[    0.003201] ftrace: allocating 18658 entries in 55 pages
[    0.046618] CPU0: update cpu_capacity 1024
[    0.046683] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.046716] [bcm2709_smp_prepare_cpus] enter
[    0.046870] Setting up static identity map for 0x8240 - 0x8274
[    0.048541] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.048868] [bcm2709_secondary_init] enter cpu:1
[    0.048910] CPU1: update cpu_capacity 1024
[    0.048916] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 389 lines cut ...]
[   10.125657] cfg80211:   (2474000 KHz - 2482000 KHz @ 8000 KHz), (N/A, 2000 mBm), (N/A)
[   10.125669] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   10.125677] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   10.125684] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   10.125690] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   10.163036] cfg80211: Current regulatory domain intersected:
[   10.163061] cfg80211:  DFS Master region: unset
[   10.163069] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[... cut ...]
[   10.174064] cfg80211:   (2457000 KHz - 2482000 KHz @ 25000 KHz), (N/A, 2000 mBm), (N/A)
[   10.174070] cfg80211:   (2474000 KHz - 2482000 KHz @ 8000 KHz), (N/A, 2000 mBm), (N/A)
[   10.174077] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   10.174085] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   10.174091] cfg80211:   (57240000 KHz - 59400000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   10.174097] cfg80211:   (59400000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

rodizio1, 2016-11-22:
Did another test on the clean image with the above patch.

Three cards were plugged in at bootup already, then plugged in the 4th card:
```
[   67.873274] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
[   67.994550] usb 1-1.3: New USB device found, idVendor=0cf3, idProduct=9271
[   67.994573] usb 1-1.3: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   67.994585] usb 1-1.3: Product: UB91C
[   67.994598] usb 1-1.3: Manufacturer: ATHEROS
[   67.994639] usb 1-1.3: SerialNumber: 12345
[   67.996104] usb 1-1.3: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   68.289608] usb 1-1.3: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   69.283254] ath9k_htc 1-1.3:1.0: ath9k_htc: Target is unresponsive
[   69.283329] ath9k_htc: Failed to initialize the device
[   69.284458] usb 1-1.3: ath9k_htc: USB layer deinitialized
```

usbmon capture:
[usbcap-with-patch-on-clean-image.zip](https://github.com/qca/open-ath9k-htc-firmware/files/607382/usbcap-with-patch-on-clean-image.zip)

olerem, 2016-11-23:
But 3 cards work stable now?

SunilMohanAdapa, 2016-11-23:
I would like share my experiences in trying to use multiple AR9271 USB Wi-Fi  dongles (TP-Link WN722N) with Olimex Olinuxino Lime 2 (OSHW).

When I tried to connect more than one USB dongle to one USB port (or even two USB ports) on the Lime2, I had issues with device showing up in lsusb but not in ifconfig.  Then I used an unpower USB hub with mostly similar results.

Suspecting a power issue, I switched to a powered USB hub.  I was able to work with at least 4 USB devices.  When I insert the USB devices one after the other every thing works.  I could even say they they work reliably.  However, after a reboot they don't get detected.  After a reboot sometimes only one device shows up.  Removing one device and inserting it back again does gets all devices working again.  There are no messages about the device detection in dmesg (such as "new high-speed USB device...").

Unfortunately, I am unable to provide logs at this moment. Please let me know if they will be useful so I can put in some time for that.

By the way, a huge thank you to @olerem and everyone else working on this project and Debian inclusion.  AR9271 based dongles are the only solution I know of for free software Wi-Fi on single board computers. This means a lot to the FreedomBox project.

olerem, 2016-11-23:
Hm... currently we start all adapters almost at once. According to wireshark capture it generate lots of traffic, especially the interrupt endpoint. By reducing number on Interrupt pipes looks like we can start at least 3 adapters at same time. @rodizio1 correct?

rodizio1, 2016-11-24:
Yes, three cards connected during boot-up worked. But I did not test very often, give me a day or two for more testing (also on the other image with the 1000Hz kernel) to really make sure it wasn't just coincidence.

rodizio1, 2016-11-27:
Did a lot more testing with both the clean image and the other with the 1000Hz Kernel. 

Reducing the interrupt pipes didn't help. It was just that I was logged in over serialport and did not use the ethernet port (which is connected over usb on the pi ...). Three cards work as long as there is nothing else happening on the USB bus. Four cards never work.

olerem, 2016-11-27:
Uff.. ok.
Currently i don't have time to do own research on it. If you or some one else can do deeper investigation it will be great. Any one who will do it, need to solder uart on the adapter, to be able to see firmware messages.

olerem, 2016-11-28:
@rodizio1 , is it your patch http://www.spinics.net/lists/linux-wireless/msg156697.html ?

rodizio1, 2016-12-03:
Not using any patches, just changed the kernel .config to use 1000Hz. So using 1000Hz should actually make it better?

Have tried soldering some wires to the Atheros Chip for serial debugging. Failed miserably, card is toast now :(

olerem, 2016-12-03:
Yes and not, 1000Hz would increase system/interrupt load.  But change behavior of timeout for this driver...
Any way, this patch went upstream. I hope it will make things better for you.
Sorry, TP-Link is hard to solder.. :(

melikyuksel, 2017-03-09:
Hey @rodizio1, did you ever find a solution to this? I'm having the same troubles with the 3 AR9271 cards + 1 ethernet adapter on the same hub. Was previously working with 2 AR9271 cards + 1 non-AR9271 card + 1 ethernet adapter, just not with 3 of the same. Getting the same errors as you.

melikyuksel, 2017-03-09:
Hmm, got it to work on this one try (probably as a fluke, but here's the summary). I booted with all 3 AR9271 cards plugged in and the USB ethernet adapter unplugged (unfortunately, you can't do that on a full-size Raspberry Pi). Logs showed that all 3 cards started fine.

53 seconds later, I unplugged one of the AR9271 cards, waited 9 seconds, and then plugged in the ethernet adapter. 8 minutes later, I plugged back in the 3rd AR9271 card that I had removed. This time, it came up. Not sure why though.

Logs:

```
[   12.399779] usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   12.411095] usb 1-1.3: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   12.418613] usb 1-1.4: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   12.420860] usbcore: registered new interface driver ath9k_htc
[   12.698557] usb 1-1.3: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   12.699463] usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   12.708310] usb 1-1.4: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   12.958874] ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[   12.959110] ath9k_htc 1-1.3:1.0: ath9k_htc: HTC initialized with 33 credits
[   12.968617] ath9k_htc 1-1.4:1.0: ath9k_htc: HTC initialized with 33 credits
[   13.399971] ath9k_htc 1-1.2:1.0: ath9k_htc: FW Version: 1.4
[   13.399985] ath9k_htc 1-1.2:1.0: FW RMW support: On
[   13.400072] ath: EEPROM regdomain: 0x809c
[   13.400076] ath: EEPROM indicates we should expect a country code
[   13.400080] ath: doing EEPROM country->regdmn map search
[   13.400084] ath: country maps to regdmn code: 0x52
[   13.400090] ath: Country alpha2 being used: CN
[   13.400092] ath: Regpair used: 0x52
[   13.402177] ath9k_htc 1-1.3:1.0: ath9k_htc: FW Version: 1.4
[   13.402186] ath9k_htc 1-1.3:1.0: FW RMW support: On
[   13.402192] ath: EEPROM regdomain: 0x60
[   13.402195] ath: EEPROM indicates we should expect a direct regpair map
[   13.402210] ath: Country alpha2 being used: 00
[   13.402213] ath: Regpair used: 0x60
[   13.406115] ath9k_htc 1-1.4:1.0: ath9k_htc: FW Version: 1.4
[   13.406165] ath9k_htc 1-1.4:1.0: FW RMW support: On
[   13.406172] ath: EEPROM regdomain: 0x833a
[   13.406175] ath: EEPROM indicates we should expect a country code
[   13.406184] ath: doing EEPROM country->regdmn map search
[   13.406190] ath: country maps to regdmn code: 0x37
[   13.406194] ath: Country alpha2 being used: GB
[   13.406196] ath: Regpair used: 0x37
[   13.600715] random: crng init done
[   13.674699] ieee80211 phy1: Atheros AR9271 Rev:1
[   13.719468] ieee80211 phy0: Atheros AR9271 Rev:1
[   13.725653] ieee80211 phy2: Atheros AR9271 Rev:1
[   15.229953] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:217084k SSFS
[   16.727251] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   18.247251] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   18.996248] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[   71.241612] usb 1-1.2: USB disconnect, device number 3
[   71.266508] ath: phy1: Failed to wakeup in 500us
[   71.395088] ath: phy1: timeout (100000 us) on reg 0x806c: 0x40ed6eda & 0x01f00000 != 0x00000000
[   71.395105] ath: phy1: RX failed to go idle in 10 ms RXSM=0x0
[   71.407455] ath: phy1: Failed to wakeup in 500us
[   71.531181] ath: phy1: timeout (100000 us) on reg 0x806c: 0x40ed6eda & 0x01f00000 != 0x00000000
[   71.531198] ath: phy1: RX failed to go idle in 10 ms RXSM=0x0
[   71.551559] usb 1-1.2: ath9k_htc: USB layer deinitialized
[   81.264473] usb 1-1.1: new high-speed USB device number 6 using dwc_otg
[   81.409631] usb 1-1.1: New USB device found, idVendor=0b95, idProduct=772b
[   81.409735] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   81.409770] usb 1-1.1: Product: AX88772B
[   81.409782] usb 1-1.1: Manufacturer: ASIX Elec. Corp.
[   81.409841] usb 1-1.1: SerialNumber: 0000D1
[   81.440727] usb 1-1.4: USB disconnect, device number 5
[   82.425769] ath: phy2: Failed to wakeup in 500us
[   82.539601] ath: phy2: timeout (100000 us) on reg 0x806c: 0x408d5ad8 & 0x01f00000 != 0x00000000
[   82.539615] ath: phy2: RX failed to go idle in 10 ms RXSM=0x0
[   82.560550] ath: phy2: Failed to wakeup in 500us
[   82.680081] ath: phy2: timeout (100000 us) on reg 0x806c: 0x408d5ad8 & 0x01f00000 != 0x00000000
[   82.680097] ath: phy2: RX failed to go idle in 10 ms RXSM=0x0
[   82.726294] usb 1-1.4: ath9k_htc: USB layer deinitialized
[   83.034228] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[... 18 lines cut ...]
[   84.204783] ieee80211 phy3: Atheros AR9271 Rev:1
[   84.358741] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   84.748039] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   84.752143] asix 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   85.402623] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  550.767674] usb 1-1.2: new high-speed USB device number 8 using dwc_otg
[  550.919790] usb 1-1.2: New USB device found, idVendor=0cf3, idProduct=9271
[  550.919817] usb 1-1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  550.919828] usb 1-1.2: Product: USB2.0 WLAN
[  550.919838] usb 1-1.2: Manufacturer: ATHEROS
[  550.919851] usb 1-1.2: SerialNumber: 12345
[  550.939194] usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[... cut ...]
[  551.919221] ath9k_htc 1-1.2:1.0: ath9k_htc: FW Version: 1.4
[  551.919281] ath9k_htc 1-1.2:1.0: FW RMW support: On
[  551.919291] ath: EEPROM regdomain: 0x809c
[  551.919294] ath: EEPROM indicates we should expect a country code
[  551.919300] ath: doing EEPROM country->regdmn map search
[  551.919304] ath: country maps to regdmn code: 0x52
[  551.919309] ath: Country alpha2 being used: CN
[  551.919313] ath: Regpair used: 0x52
[  551.945790] ieee80211 phy4: Atheros AR9271 Rev:1
[  553.179864] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
```

melikyuksel, 2017-03-10:
@rodizio1 Disregard that, I hadn't read carefully once again. I'll apply the patch and see what happens.

melikyuksel, 2017-03-12:
I applied the patch from the link and confirmed it was patched, but the problem persists in the exact same way.

```
[   56.784735] usb 1-1.4: Service connection timeout for: 256
[   56.784813] ath9k_htc 1-1.4:1.0: ath9k_htc: Unable to initialize HTC services
[   56.784912] ath9k_htc: Failed to initialize the device
[   56.788045] usb 1-1.4: ath9k_htc: USB layer deinitialized
```

The patch is supposed to make it timeout after 1 second instead of 0.1 seconds, but judging by the log, there's no change in the time before the the "Service connection timeout" occurs, which is around 300 ms. I changed the code to 3 seconds, 5 seconds, and 10 seconds, and it has no effect at all.

melikyuksel, 2017-03-12:
Got it working! Did a test with the patched ath9k_htc module on a desktop running Debian, and it didn't have any problems with 3, 4, or even 5 AR9271 devices plugged in. So I went back to the Raspberry Pi and did a `rpi-upgrade`. This moved me from January's `4.9.13+` kernel to the version which I believe is included in this month's new release of Raspbian, `4.4.50+`. After reboot, all 3 AR9271 devices worked!

rodizio1, 2017-04-05:
Sorry for my late reply melikyuksel, didn't realize you were posting here.

Well, what you describe is exactly what I'm seeing. It appears to be more or less random and also being dependent on other stuff using the USB bus.

Three cards is 99% now for me, as long as I make sure no other stuff is running during card initialization and no network cable is plugged. Oh and LED blinking disabled.

Maybe we should open another issue on the Raspberry Linux kernel Github.

melikyuksel, 2017-04-15:
@rodizio1 Sounds like a good idea. This does seem to be limited to Raspberry Pi kernels; I assume Kali Linux for Raspberry Pi also uses the same kernel as Raspbian, since it has the problem as well. According to the mailing list, it used to be a problem for all of the Linux kernel, but was fixed with that patch above; but applying that patch doesn't fix it for Raspberry Pi, so it must depend on some other new development that the Pi kernel doesn't have yet... or it's a hardware thing. I'll try it out with Arch Linux on the Pi and see what happens. If not, I'll submit an issue on raspberrypi/linux, unless you want to be proactive. :)

sumeetkpatel, 2017-05-24:
I am also struggling with 4X antenna setups using TPLink W722N. I am trying to create a device that I can reboot and have all the cards come up reliably on kernel 4.x (3.x works oddly enough), I am simply at a loss at what is happening so started collecting details about whats happening.

I am not seeing the exact same errors as OP, but rather am having trouble initializing the devices at startup altogether, in the scenarios where all 4 cards work, we haven't noticed any problems (yet) during monitor mode. Perhaps it could be related?

I can confirm that power hasn't been an issue at all, as running powered USB hubs, using multi-meters and myriads of hub setups has had no hugely positive results and many times I have had the four cards working all on single hub from a single USB port without additional power.

I would note here that our experience with OTG USB such as the one on the Raspberry Pi (both 1 and 2) provoked the issue so badly that we moved to devices that do not use OTG such as the OrangePi Lite as we suspected the OTG just cant handle so many USB devices and especially in this manner.

I would note also that we have had similar issues on other devices, even with a proper USB hub, such as the new ASUS Tinkerboard, OrangePi Lite, and OrangePi Zero.

Funnily enough, we have no issues with the x86 based Intel Compute Stick, but is an expensive device.

We have had less issues on ARM based systems with the Ralink based Mediatek (mt7601u) setup, but is still problematic and unrelated.

So using the OrangePi Lite we do not have issues using 3.x based Armbian or DietPi based distro.

I am including information on my current setup in development, the OrangePiLite / Armbian nightly Kernel 4.11:

* Disconnected at boot-time:
    * Having the single hub USB removed, booting, then connecting the hub USB, all 4 cards come up! Great!, but that's not good in my scenario as we wont be around to plug/unplug at boot-time. But also proved 4 are possible and can be put into monitor mode successfully without errors, but I'll continue to monitor that scenario.

* Connected at boot-time:
    * Having all 4 cards connected, once boot is complete, only 2 cards are usually active but have seen results varying from 0-4 cards working, typically 2.
    * I have attached various logs and output of commands for 2, 3 and 4 cards operational on boot (with all 4 cards actually plugged in).
    * You can see clearly that "lsusb" is reporting all 4 cards and in "dmesg" all 4 cards request the firmware but not all of them actually respond after that, you can also see the sysfs is missing the ieee80211 folder in the devices that are not operational.
    * The attached logs are for OrangePi Lite using the latest nightly updated.

I have not yet tried compiling the driver yet with debug enabled.

venkatesh-kuppan, 2017-10-20:
I am trying to interface TL-WN722N with Angstrom Image on a Cyclone V SoC. I have built the module as static and initialized the firmware. I am using the following kernel version : 4.1.33-ltsi-altera . 
I observe the following error now : 

```
[   14.568484] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   14.820407] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[   15.852173] ath: phy0: Mac Chip Rev 0x0f.3 is not supported by this driver
[   15.859064] ath: phy0: Unable to initialize hardware; initialization status: -95
[   15.866512] ath: phy0: Unable to initialize hardware; initialization status: -95
[   15.873973] ath9k_htc: Failed to initialize the device
```

I had a read over the linux drivers for ath9k_htc on the official website for TP-Link(the manufacture of wifi dongle), and it says that the `ath9k` drivers are officially only available for the version 2 of the device. The TL-WN722N is a version 1 device as per the details on the website. 

Has anyone faced such an issue. 
Note: The version can be identified by reading the FCC ID printed on the device, it suffixes `v2` specifically for a version 2 device and none for version 1 devices.

melikyuksel, 2018-04-23:
I think this can be tentatively be closed, now that this was fixed at the end of https://github.com/raspberrypi/linux/issues/2023. Or we could keep it open until the kernel that fixes it becomes stable so `rpi-update` will not be necessary, or until the the `/boot/cmdline.txt` change is not needed (ever)?

olerem, 2018-04-23:
Do i see it correctly, the fix is related to RPi (or dw) USB host controller and not to atheros driver or firmware?

dagf2101, 2019-02-06:
I found some sort of workaround to my multiple adapter problem,

Using kernel 4.19 and two adapters plugged in, I had one of the two adapters failing maybe once every 5-10 reboots.

Once in that state, I tried so many things to completly restart the card or pc or turn off USB power without success, but finally managed to do it using a realtimeclock wake alarm.

It may be not implemented on every systems but on my NUC pc I do the following:
#!/bin/bash
echo 0 > /sys/class/rtc/rtc0/wakealarm"
echo `date '+%s' -d '+ 10 seconds'` > /sys/class/rtc/rtc0/wakealarm
shutdown -h now

The system will shut down and in my case TURN OFF THE USB POWER, and then restart after 10 seconds.
Just make a script at boot to automate this until all the adapters are working.

I know this is not ideal but im my case that is a acceptable workaround to make the device boot in a working state.