qca/open-ath9k-htc-firmware #115: Firmware debug

doru91, 2016-11-25:
Hello, everyone

I'm trying to print the value of an internal firmware variable. I was wondering if there is a dmesg equivalent mechanism.

What is the easiest method for the above issue?

Thank you,
Doru

olerem, 2016-11-25:
you need to grab it directly from the chip. Search for UART on this page https://wikidevi.com/wiki/Atheros_AR9271

doru91, 2016-11-25:
It seems that ALFA_Network_AWUS036NHA has accesible UART. 

I am not a hardware person, so could you please indicate me what type of connections I need for UART? I was looking at the top board picture from [1] but I can't figure out what type of hardware I need for UART.

I assume that you already have a debugging board with UART connections, maybe you could send me a picture/description of it.

[1] https://wikidevi.com/w/images/c/cd/ALFA_Network_AWUS036NHA_newRevision_%28AR9271L%29_board_top.jpg

olerem, 2016-11-25:
On the top right side of the chip on the beard are two points: RXD1 and TXD1, this is uart. you need to solder 3 cables: ground, RX and TX. And attach them to some USB to TTL UART adapter. For example:
http://www.ebay.de/itm/USB-to-TTL-Konverter-Arduino-RS232-3-3V-5V-PL2303-UART-Adapter-Seriell-Converter-/291898514447?hash=item43f681dc0f:g:P94AAOSwa-dWoyr7

olerem, 2016-11-25:
here is my HW https://plus.google.com/102032716864870215256/posts/aCua2D6729J
make sure uart adapter is set to 3.3Volt and not to 5 Volt

doru91, 2016-11-25:
Thanks a lot!

doru91, 2016-11-30:
I soldered GND, TX and RX cables between the the UART adapter and the Wi-Fi board.

How did you solder the 3.3 PIN? Did you use a source from the Wi-Fi board?

olerem, 2016-11-30:
It is not needed. at least i never used it.

olerem, 2016-11-30:
But i have a bit different adapter.

erikarn, 2016-11-30:
You shouldn't need the 3.3v line. Just TX/RX/gnd

-a

doru91, 2017-02-18:
Right, I make it work without the 3.3v line.

Thanks!