qca/open-ath9k-htc-firmware #123: CCA carrier sense and energy detect thresholds?

rodizio1, 2017-01-22:
I noticed AR9287 cards seem to be much more "cautious" when injecting traffic, i.e. if they refuse to inject if there is some other (I think non-Wifi) signal present, while AR9271 cards only refuse to inject when there is a tranmitter nearby.

I found some reference in 802.11 standards to the thresholds being -82dbm for carrier sense (i.e. other wifi traffic) and -62dbm for energy detect (non-wifi signals).

My feeling is somehow, AR9287 cards also use the -82dbm value for non-wifi signals (?)

Looked through the drivers and found something in ar9002_phy.h:
```
#define AR_PHY_CCA_NOM_VAL_5416_2GHZ            -90
#define AR_PHY_CCA_NOM_VAL_5416_5GHZ            -100
#define AR_PHY_CCA_MIN_GOOD_VAL_5416_2GHZ     -100
#define AR_PHY_CCA_MIN_GOOD_VAL_5416_5GHZ     -110
#define AR_PHY_CCA_MAX_GOOD_VAL_5416_2GHZ     -80
#define AR_PHY_CCA_MAX_GOOD_VAL_5416_5GHZ     -90
#define AR_PHY_CCA_NOM_VAL_9280_2GHZ         -112
#define AR_PHY_CCA_NOM_VAL_9280_5GHZ         -112
#define AR_PHY_CCA_MIN_GOOD_VAL_9280_2GHZ  -127
#define AR_PHY_CCA_MIN_GOOD_VAL_9280_5GHZ  -122
#define AR_PHY_CCA_MAX_GOOD_VAL_9280_2GHZ  -97
#define AR_PHY_CCA_MAX_GOOD_VAL_9280_5GHZ  -102
#define AR_PHY_CCA_NOM_VAL_9285_2GHZ           -118
#define AR_PHY_CCA_MIN_GOOD_VAL_9285_2GHZ    -127
#define AR_PHY_CCA_MAX_GOOD_VAL_9285_2GHZ    -108
#define AR_PHY_CCA_NOM_VAL_9271_2GHZ             -118
#define AR_PHY_CCA_MIN_GOOD_VAL_9271_2GHZ      -127
#define AR_PHY_CCA_MAX_GOOD_VAL_9271_2GHZ      -116
#define AR_PHY_CCA_NOM_VAL_9287_2GHZ           -112
#define AR_PHY_CCA_MIN_GOOD_VAL_9287_2GHZ    -127
#define AR_PHY_CCA_MAX_GOOD_VAL_9287_2GHZ    -97
```

Ist this the right place to experiment with those values? Somehow they seem too low, are there offsets or are these half-db values maybe?

erikarn, 2017-01-22:
hi,

The values have to do with the PHY behaviour and calibration behaviour.
They're not exact dBm values.

It should always be -62dB for energy detect; that's programmed elsewhere.

I .. hm, this reminds me of some outstanding PHY questions. Lemme go ask.

-adrian

erikarn, 2017-01-24:
Hiya,

ok. So, the TL;DR is:

* the calibrated NF that you read is hopefully the "thermal" noise floor
* everything else in the PHY, CCA included, is relative to the calibrated NF
* so if the calibrated NF value is too high, because of a noise spur, or it
did it at the "wrong" time, a lot of the radio can go "deaf"
* that's why there's a lot of filters around what NF calibration values are
read and programmed in!

Those min/max/good values aren't in dBm. There's no explicit absolute dBm
value anywhere. Yes, a lot of drivers "lie". :-)

-adrian

rodizio1, 2017-01-27:
Thanks. So this all is tied to the ANI functionality (which seems to be dynamic) and not just fixed values?

How can I further debug this? AR9287 cards become pretty much unusable as soon as there is some other faint signal around.