qca/open-ath9k-htc-firmware #126: Driver stops receiving packets after several seconds to a several hours on monitor mode

melikyuksel, 2017-02-24:
Both version 1.3 and 1.4 seem to have the issue. After anywhere from 20 seconds to 3 hours or so of being in monitor mode, the interface stops receiving packets. The `ifconfig` count stops increasing, and the `debug_fs` numbers also become static. The only way to get it to receive again is to change the channel (changing to another channel and then immediately switching back works). It doesn't seem to be able to stay on one channel for very long.

I've tried this with several Atheros AR9271 chipset cards, like the ALFA AWUS036NHA and TL-WN722N. They all have the same issue. I've tried it on Kali, Raspbian, and Arch, and I've gotten identical behavior. I've tried a powered USB hub, an unpowered hub, and no hub at all. I've tried it on a MacBook, a Mac Mini, a Raspberry Pi, and a Pi Zero. I've tried it with all sorts of different antenna. I've tried power supplies of various amperages. Power save is off in both module parameters and in `iw`.

olerem, 2017-02-24:
@erikarn , hm... should we do some periodic recalibration?

erikarn, 2017-02-24:
lol uhm, yes? :)

Are you telling me ath9k_htc is not doing any periodic calibration either
in firmware or in the host side? I expected the host side is doing some
stuff

-a

olerem, 2017-02-24:
Not 100% sure if we are not doing. But is sounds like not, at least not for monitoring mode.

erikarn, 2017-02-24:
Yes, we should be, even in monitor mode.

-adrian

melikyuksel, 2017-02-24:
This isn't exclusive to the open firmware btw, also happens in the one that comes with the kernel.

rodizio1, 2017-02-24:
Interesting. I have had similar effects when being in monitor mode _and_ having set fcsfail, i.e. frames with wrong checksum are also forwarded.

I could reproduce this by using my EZ-Wifibroadcast distribution, enabling fcsfail in monitor mode and then running it near sensitivity limit, so that there are something like 50% of wrong checksum frames being received. Then sometimes, it would just stop receiving packets, like you described, no more ifconfig packets/bytes updates.

Never tried to switch channel though when that happened. But I remember ifconfig down/up didn't work.

Without FCS Fail, this never happened, have been running these cards for several days in monitor mode constantly receiving ~700 1024 byte frames per second.

olerem, 2017-02-24:
@melikyuksel  can check if fcsfail ist set on you system?

melikyuksel, 2017-02-24:
@olerem The `fcsfail` flag is not set. Just to check, I turned it on for both AR9271 cards, and both still eventually stopped within 30 minutes (lots of traffic at the moment). I turned it off, and both again stopped receiving packets again within the next hour.

I'm trying all this on fresh installations, and that flag is turned off by default when using both `iw` and `airmon-ng`.

rodizio1, 2017-02-24:
Not sure what the default is, I explicitly set it with `iw dev wlan0 set monitor none`. 

When I use `iw dev wlan0 set monitor fcsfail` it sooner or later stops receiving frames when too much "garbage" frames come in.

melikyuksel, 2017-02-24:
I suspect that your cards would also stop receiving frames irregardless of `fscfail` if you were in a crowded enough area. I might be wrong and you might already be though. I've noticed that the more packets the card is seeing, the sooner it fails. Turning on `fcsfail` probably just increases the number of processed packets and leads to failure. In my case, I'm in such a crowded area that the card fails even without setting that flag on.

olerem, 2017-02-25:
in this case, after monitoring is stalled, ifconfig will show you identical or similar amount of packages

olerem, 2017-02-25:
I started monitore test:
ifconfig wlxa0f3c1187553 up
airmon-ng start wlxa0f3c1187553 6
tcpdump -ni mon0

Currently it is running one hour.

melikyuksel, 2017-02-25:
I left two running overnight as well (and they didn't stall, because they never stall overnight for some reason), and one just stalled at 5,138,158 RX packets, while the other is still going at 6,336,815 RX packets so far. Is there some way to debug the situation when they stall? The ath9k_htc debug system itself show nothing.

rodizio1, 2017-02-25:
> I suspect that your cards would also stop receiving frames irregardless of fscfail if you were in a crowded enough area. I might be wrong and you might already be though.

I have tested it with detached antennas on the receiving wifi sticks (PCB Antennas on 722N also disabled) all they could hear was the packets coming from my transmitting wifi sticks (i.e. the videostream).

> I've noticed that the more packets the card is seeing, the sooner it fails. Turning on fcsfail probably just increases the number of processed packets and leads to failure. In my case, I'm in such a crowded area that the card fails even without setting that flag on.

This was not the case with my tests, the transmitting stick sends something around 700 frames per second. When being near, lost frames or frames with failed FCS are almost zero, maybe one per 5 minutes or so. Then it runs for days. 
When going to the edge of reception, there are actually _less_ frames being received as some are completely lost.  I also noticed that this 'stall' happens quicker when there are more failed FCS frames coming in.

So my theory so far is, that it's not the actual number of frames that causes this (i.e. "overloading" the card when there are too many frames). I think it's that some special combination of garbage data causes the card to crash (some invalid combinations of whatever fields in the IEEE802.11 header or maybe something in the PLCP pre-amble or header, wrong length field that causes a read past where it shouldn't read, something like that).

This would explain that it can be reproduced quicker when there are more failed FCS frames, the probability that some frame contains "garbage the cards don't like" is higher.

However, this explanation of course doesn't line up with your results with fcs fail not enabled. Still, I feel quite confident that this does never happen with my EZ-Wifibroadcast distribution, maybe you could test it with that? This way we'd have completely the same setup, only variable left would be local circumstances like other wifi networks.

You can download it from here: https://github.com/bortek/EZ-WifiBroadcast/wiki, write it to an SDcard, edit wifibroadcast-1.txt on the boot partition (only FREQ=XXXX needs to be changed) put it in the Pi and let it run. SSH login is standard l:root p:raspberry. If you want local console access, switch over to tty10.

melikyuksel, 2017-02-25:
@rodizio1 Sure, I'll test it with EZ-Wifibroadcast. Do I just boot it up and put the interface in monitor mode, or do I need a second device like it says on the page?

rodizio1, 2017-02-25:
No second device needed, just write the image onto an SDcard and boot it up. Cards will automatically be in monitor mode on channel 13 after bootup. You can use up to three wifi sticks.

erikarn, 2017-02-25:
Hiya,

can you turn on NFCAL debugging and see what it's doing?

-adrian

melikyuksel, 2017-02-25:
@erikarn How is that done?

erikarn, 2017-02-25:
https://wireless.wiki.kernel.org/en/users/Drivers/ath9k/debug

compile ath9k/ath9k_htc with debugging, then enable the CALIBRATE bit as
mentioned there, and check dmesg

it should be doing NF cal every 30 seconds or so.

-adrian

rodizio1, 2017-02-25:
It doesn't do periodic noisefloor calibration. Only when changing channels.

melikyuksel, 2017-02-25:
Just got the EZ-Wifibroadcast image running for one card, and on the other card enabled the calibrate bit. No messages in dmesg even with the bit set though, and the packets haven't stalled on either yet.

erikarn, 2017-02-26:
hiya,

ok, let's figure out how/why behind doing periodic calibrations. That
sounds like a bit of a bug.

-adrian

melikyuksel, 2017-02-26:
So, the EZ-Wifibroadcast image card has stalled with the `fcsfail` option untouched (which I assume means it's off). The card with calibrate bit enabled has not stalled yet as of 12 hours later, and there are no messages in `dmesg` at all about it. Are they really supposed to be in `dmesg`, or maybe somewhere else?

erikarn, 2017-02-26:
they're supposed to be in 'dmesg'. type dmesg, don't look at console output.

If it only is logged during NIC startup then yeah, let's figure out how to
fix that!

-adrian

melikyuksel, 2017-02-26:
I'm definitely not seeing any periodic messages in dmesg from any of the cards, but I don't think there's anything on startup either. @rodizio1 Do you know what the message is supposed to look like? It's strange because `/sys/kernel/debug/ieee80211/phy1/ath9k_htc/debug` does contain `0x00000008` (the CALIBRATE bit).

rodizio1, 2017-02-26:
>So, the EZ-Wifibroadcast image card has stalled with the fcsfail option untouched (which I assume means it's off).

Well, crap :) That makes my "bad content frames" theory invalid or we have two different issues here. 

Yes, it doesn't forward fcsfail frames by default.  I'll make something that allows to inject a lot of frames for testing to verify your "too much frames" theory.

This is what I get with debug=0xffffffff module parameter, NF calibration is done, but only on ifconfig up or channel change:

Note, the lines starting with "ATH:" (in uppercase) are printk's that I added myself to the source to see which function the driver calls.

```
[    4.876551] ath9k_htc 1-1.5:1.0: ath9k_htc: HTC initialized with 33 credits
[    4.890704] ath: phy1: serialize_regmode is 0
[    4.890721] ath: phy1: UNDEFINED -> AWAKE
[    4.898669] ath: phy1: Reading from EEPROM, not flash
[    5.039843] ath: phy1: Read Magic = 0xA55A
[    5.039857] ath: phy1: need_swap = False
[    5.039874] ath: phy1: Eeprom VER: 14, REV: 13
[    5.039880] ath: phy1: Initialize ANI
[    5.039886] ATH: ath9k_ani_restart function
[    5.039892] ath: phy1: Enable MIB counters
[    5.041696] ath: phy1: Setup TX queue: 9
[    5.041704] ATH: ath9k_hw_set_txq_props
[    5.041710] ath: phy1: Set queue properties for: 9
[    5.041717] ath: phy1: Setup TX queue: 8
[    5.041721] ATH: ath9k_hw_set_txq_props
[    5.041726] ath: phy1: Set queue properties for: 8
[    5.041731] ath: phy1: Setup TX queue: 1
[    5.041736] ATH: ath9k_hw_set_txq_props
[    5.041740] ath: phy1: Set queue properties for: 1
[    5.041747] ath: phy1: Setup TX queue: 0
[    5.041750] ATH: ath9k_hw_set_txq_props
[    5.041755] ath: phy1: Set queue properties for: 0
[    5.041760] ath: phy1: Setup TX queue: 2
[    5.041764] ATH: ath9k_hw_set_txq_props
[    5.041769] ath: phy1: Set queue properties for: 2
[    5.041774] ath: phy1: Setup TX queue: 3
[    5.041779] ATH: ath9k_hw_set_txq_props
[    5.041784] ath: phy1: Set queue properties for: 3
[    5.176018] gpiomem-bcm2835 3f200000.gpiomem: gpiomem device opened.
[    5.210360] systemd[1]: Started wbcconfig.
[    5.211821] systemd[1]: Startup finished in 3.157s (kernel) + 2.053s (userspace) = 5.211s.
[    5.322941] ath: phy1: TX streams 1, RX streams: 1
[    5.323782] ath9k_htc 1-1.5:1.0: ath9k_htc: FW Version: 1.4
[    5.323800] ath9k_htc 1-1.5:1.0: FW RMW support: On
[    5.323809] ath: EEPROM regdomain: 0x809c
[    5.323814] ath: EEPROM indicates we should expect a country code
[    5.323821] ath: doing EEPROM country->regdmn map search
[    5.323826] ath: country maps to regdmn code: 0x52
[    5.323832] ath: Country alpha2 being used: CN
[    5.323838] ath: Regpair used: 0x52
[    5.431209] ath: phy1: WMI:1, BCN:2, CAB:3, UAPSD:4, MGMT:5, BE:6, BK:7, VI:8, VO:9
[    5.431228] ieee80211 phy1: Atheros AR9271 Rev:1
[    5.484051] cfg80211: Regulatory domain changed to country: CN
[    5.484076] cfg80211:  DFS Master region: unset
[    5.484085] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    5.484096] cfg80211:   (2302000 KHz - 2742000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[    5.484104] cfg80211:   (4910000 KHz - 5835000 KHz @ 160000 KHz), (N/A, 3000 mBm), (N/A)
[    7.979701] cfg80211: Regulatory domain changed to country: DE
[    7.979719] cfg80211:  DFS Master region: unset
[    7.979724] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    7.979734] cfg80211:   (2302000 KHz - 2742000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[    7.979742] cfg80211:   (4910000 KHz - 5835000 KHz @ 160000 KHz), (N/A, 3000 mBm), (N/A)
[    8.001992] ath9k_htc 1-1.5:1.0 c46e1f21bbf1: renamed from wlan0
[    8.605963] ath: phy1: Starting driver with initial channel: 2312 MHz
[    9.580319] ath: phy1: PDADC (0,   0): a280 04030201
[    9.580336] ath: phy1: PDADC: Chain 0 | PDADC   0 Value   1 | PDADC   1 Value   2 | PDADC   2 Value   3 | PDADC   3 Value   4 |
[    9.580344] ath: phy1: PDADC (0,   0): a284 08070605
[    9.580352] ath: phy1: PDADC: Chain 0 | PDADC   4 Value   5 | PDADC   5 Value   6 | PDADC   6 Value   7 | PDADC   7 Value   8 |
[    9.580359] ath: phy1: PDADC (0,   0): a288 0d0c0a09
[... 644 lines cut ...]
[   11.573556] ath: phy1: tx ok 0x0 err 0x0 desc 0x10f eol 0x10f urn 0x0
[   11.574554] ath: phy1: Reset TXQ, inactive queue: 4
[   11.574560] ath: phy1: Reset TXQ, inactive queue: 5
[   11.574565] ath: phy1: Reset TXQ, inactive queue: 6
[   11.574570] ath: phy1: Reset TXQ, inactive queue: 7
[   11.574575] ath: phy1: Reset TX queue: 8
[   11.578554] ath: phy1: tx ok 0x0 err 0x0 desc 0x10f eol 0x10f urn 0x0
[   11.579554] ath: phy1: Reset TX queue: 9
[   11.582556] ath: phy1: tx ok 0x0 err 0x0 desc 0x10f eol 0x10f urn 0x0
[   11.584555] ath: phy1: ver 320.1 opmode 6 chan 2484 Mhz
[   11.592681] ath: phy1: ah->misc_mode 0x4
[   11.656582] ath: phy1: enabling ADC Gain Calibration
[   11.656595] ath: phy1: enabling ADC DC Calibration
[   11.656600] ath: phy1: enabling IQ Calibration
[   11.658584] ath: phy1: starting ADC Gain Calibration
[   11.663579] ATH: ath9k_htc_opmode_init
[   11.665585] ATH: ath9k_htc_calcrxfilter
[   11.665596] ATH:ath9k_hw_setrxfilter
[   11.669567] ath: phy1: Enable MIB counters
[   11.671565] ath: phy1: **** ofdmlevel 3=>3, rssi=0[lo=7 hi=40]
[   11.671574] ath: phy1: **** ccklevel 2=>2, rssi=0[lo=7 hi=40]
[   11.671579] ATH: ath9k_ani_restart function
[   11.671584] ATH: ath9k_ani_restart function after if (!ah->curchan)

```

melikyuksel, 2017-02-26:
That's strange, I wasn't even getting these messages when I had 0xffffffff set. I have the kernel compiled for debugging ath9k_htc and set the bit and everything.

rodizio1, 2017-02-26:
I have these wireless related debugging settings in the kernel .config file:

```
CONFIG_CFG80211_DEBUGFS=y
CONFIG_MAC80211_DEBUGFS=y
CONFIG_ATH_DEBUG=y
CONFIG_ATH9K_DEBUGFS=y
CONFIG_ATH9K_HTC_DEBUGFS=y
```

melikyuksel, 2017-02-26:
Thanks, turns out I neglected to enable `CONFIG_ATH_DEBUG` the first time around. Now I got it recompiled, and I can also report that it's not doing any periodic calibrations. Here's the output for one of the cards when I switched the channel to 2 from channel 1 after it stalled:

```
[ 2909.583318] ath: phy0: NF calibrated [ctl] [chain 0] is -112
[ 2909.583333] ath: phy0: NF[0] (-112) > MAX (-116), correcting to MAX
[ 2909.583340] ath: phy0: NF calibrated [ext] [chain 0] is -110
[ 2909.583382] ath: phy0: NF[3] (-110) > MAX (-116), correcting to MAX
[ 2909.752322] ath: phy0: enabling ADC Gain Calibration
[ 2909.752335] ath: phy0: enabling ADC DC Calibration
[ 2909.752340] ath: phy0: enabling IQ Calibration
[ 2909.754301] ath: phy0: starting ADC Gain Calibration
```

Here it is for the other card after it similarly froze and I manually changed the channel to 2:

```
[ 2963.387943] ath: phy1: NF calibrated [ctl] [chain 0] is -107
[ 2963.387965] ath: phy1: NF[0] (-107) > MAX (-116), correcting to MAX
[ 2963.387974] ath: phy1: NF calibrated [ext] [chain 0] is -106
[ 2963.387987] ath: phy1: NF[3] (-106) > MAX (-116), correcting to MAX
[ 2963.568711] ath: phy1: enabling ADC Gain Calibration
[ 2963.568724] ath: phy1: enabling ADC DC Calibration
[ 2963.568732] ath: phy1: enabling IQ Calibration
[ 2963.570441] ath: phy1: starting ADC Gain Calibration
```

And after switching back to channel 1:
```
[ 2963.791457] ath: phy1: NF calibrated [ctl] [chain 0] is -105
[ 2963.791472] ath: phy1: NF[0] (-105) > MAX (-116), correcting to MAX
[ 2963.791479] ath: phy1: NF calibrated [ext] [chain 0] is -108
[ 2963.791523] ath: phy1: NF[3] (-108) > MAX (-116), correcting to MAX
[ 2963.967462] ath: phy1: enabling ADC Gain Calibration
[ 2963.967474] ath: phy1: enabling ADC DC Calibration
[ 2963.967483] ath: phy1: enabling IQ Calibration
[ 2963.969444] ath: phy1: starting ADC Gain Calibration
```

There are no new messages in `dmesg` after it starts stalling, nor when it is performing normally.

melikyuksel, 2017-02-26:
@rodizio1 Here's something to try: I wrote a script to switch through all the channels 1–11 every 10 minutes and return back to channel 1 for another 10 minutes, and found that it begins stalling a lot faster after this. It stops stalling of course when the next channel hopping session begins. My script also gets the packets seen in ifconfig and prints something out if it doesn't change within 3 seconds, so that's how I know it's stalled.

erikarn, 2017-02-26:
the bug is that we should be doing periodic nfcal and other calibrations or
the nics will slowly go deaf.

-a

melikyuksel, 2017-02-27:
Right. Apart from the conspicuous lack of periodic calibration messages, the only things I've noticed is this message occasionally in `dmesg`:

```
[ 5743.651260] ath: phy0: Timeout waiting for WMI command: WMI_REG_RMW_CMDID
[ 5743.651314] ath: phy0: REGISTER RMW FAILED:(0x4048, -110)
```

and the extreme increase in the frequency of the stalling when channels are switched frequently. If I run a script to change the channel once every 10 minutes, the card begins to stall within minutes or seconds of being set to a channel.

olerem, 2017-02-28:
@melikyuksel , it has nothing to do with number of packages. My adapter was running for some days now, and it is still working. More noise in the air will force the adapter to increase the noise floor level on rx line. The noise is not always at same level, it can have spikes. So if adapter will apply the filter at the level of this spike, then we will get deaf.
The ath9k code has logic for periodic calibration. ath9k-htc don't have it. Some one should make this code common, between ath9k and ath9k-htc.

melikyuksel, 2017-03-02:
@olerem Thanks for the explanation, it makes a lot of sense (although not quite sure why switching channels often triggers a very quick deafening... maybe more noise is introduced during the switching process?). Is there anything one could do to add the calibration to the -htc themselves at the moment?

olerem, 2017-03-05:
@melikyuksel , ... "Some one should make this code common, between ath9k and ath9k-htc.", most probably i will not find time to do this. So every one is welcome to contribute ... it is open source.
The patch will look probably like this:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=67dc74f15f147b9f88702de2952d2951e3e000ec

rodizio1, 2017-03-09:
@melikyuksel Those crashes/issues you see while switching channels is a different issue. I also have these problems, something is buggy there, also happens when switching to promiscous mode sometimes.

@olerem/erikarn: How can you guys be sure that it's the missing periodic noise floor calibration that is causing the issue? Why can I reproduce this without any noise-spikes being present, just by receiving a lot of bad-FCS frames? Different issue?

melikyuksel, 2017-03-09:
@rodizio1 True. I ended up writing a script to check for ath9k_htc interfaces, check if they're on monitor mode, and then check the RX packages every 5 seconds. If the packets stop increasing, it just switches the channel and then switches it back. This seems to have "solved" these issues, but now I'm having the issue when multiple ath9k_htc cards don't start that you were having in the other thread. That's a whole another matter in itself...

erikarn, 2017-03-09:
It looks like HTC NICs do calibration: ath9k_htc_start_ani() will start the
ani work, and ath9k_htc_ani_work() will do the actual ANI, shortcal,
longcal work.

The trouble(!) is, the ani_work is only added in certain circumstances,
which is "technically" correct for ANI, but wrong for shortcal/longcal.

Eg - in ath9k_htc_add_interface(), it's only started if it's an AP, and ANI
is enabled. For STA mode in ath9k_htc_bss_info_changed(), it's only started
if it's associated, and number of associated vifs is 1.

For monitor mode, it's not doing ANI, which means its not doing periodic
calibration.

So as a test, let's just enable it always for monitor mode.

diff --git a/drivers/net/wireless/ath/ath9k/htc_drv_main.c
b/drivers/net/wireless/ath/ath9k/htc_drv_main.c

index a553c91..a4f84c2 100644

--- a/drivers/net/wireless/ath/ath9k/htc_drv_main.c
+++ b/drivers/net/wireless/ath/ath9k/htc_drv_main.c
@@ -1104,6 +1104,11 @@ static int ath9k_htc_add_interface(struct
ieee80211_hw *hw,
                ath9k_htc_start_ani(priv);
        }

+       if (vif->type == NL80211_IFTYPE_MONITOR) {
+               printk("%s: starting ANI, monitor mode\n", __func__);
+               ath9k_htc_start_ani(priv);
+       }
+
        ath_dbg(common, CONFIG, "Attach a VIF of type: %d at idx: %d\n",
                vif->type, avp->index);

And I'll go look at tidying this code up to always run the ANI routine in
hostap/STA/monitor/mesh mode so periodic calibration runs, then just skips
ANI if ANI is configured as disabled. This whole "don't do ANI" stuff is
conflating with the "do periodic calibration" stuff, and that's why your
NICs go deaf. :-)

-adrian

melikyuksel, 2017-03-12:
@erikarn I would love to try this patch on the Raspberry Pi, but I can't seem to recompile the ath9k module on there without somehow messing up its ability to recognize more than 2 AR9271 devices at the same time (seen in the other issue thread). I will be trying this, however, on Debian!

erikarn, 2017-03-12:
hiya,

please do test it out. I'll get linux-next going again on something
laptop-y next week so I can fully figure out what's going on with all of
this calibration nonsense. :)

-adrian

melikyuksel, 2017-03-13:
@erikarn I seem to have had the first surrender, where one of the cards stopped after 50 minutes (with channel hopping being done every 10 minutes). But where is that `printk` message supposed to show up? I checked dmesg and did not see it there, but I modified some of the messages in hif_usb.c at the same time, and I *do* see those. Those use the `dev_info` function, apparently:

```
dev_info(&hif_dev->udev->dev, "ath9k_htc: Transferred FW: %s, size: %ld\n",
                 hif_dev->fw_name, (unsigned long) hif_dev->fw_size);
```

erikarn, 2017-03-13:
It's supposed to show up when the interface is created. If it's not
printing it out, then it's not starting.

Which kernel version are you using? Does ti really not show up in 'dmesg' ?

-adrian

melikyuksel, 2017-03-17:
@erikarn This is 4.4.50. It doesn't show up in dmesg, but expected lines from the other files do show up. Just not this one.

erikarn, 2017-03-17:
ok, need to figure out why that is. Can someone else help out enabling the
ANI timer like I have in that patch? It should make things work.

-adrian

rodizio1, 2017-03-20:
I have added the patch now to Raspberry Kernel 4.4.11. It seems the "if (vif->type ..." condition does not match. Commented out the if condition and then I get the "ath9k_htc_add_interface: starting ANI, monitor mode" debug line in dmesg.

erikarn, 2017-03-20:
ok, does that fix it?

also, can you printk the vif->type value? What interface type are you
creating?

-adrian

melikyuksel, 2017-03-27:
@erikarn Had the chance to test it. The `vif->type` value prints out as 2, which, if [this](http://lxr.free-electrons.com/source/include/linux/nl80211.h?v=2.6.24) is where it's getting it from, means `NL80211_IFTYPE_STATION` instead. But it seems this part of the code only runs when the device is first plugged in (by default in station mode), and not again even when we bring the interface up and later when we put it into monitor mode.

rodizio1, 2017-04-05:
Yes, I can confirm this. dmesg debug log only appears once after plugging the card. 

Is this sufficient, i.e. will the chip still do noise floor calibration after it has been "ifconfig upped/downed" and put into monitor mode?

erikarn, 2017-04-05:
hi,

ok, we need to figure out how to get it to start ANI whenever the interface
is created as well as down/up. It should always be doing period noise
calibration when the NIC is up.

-adrian

occam25, 2017-05-08:
Did you manage to solve this issue? I am having the same problem with COMPEX WLE350NX-7B mPCIe module (AR9580) and kernel 3.2.70. I tried all your suggestions with same results.. 
By now I solve this issue by detecting that the interface is stalled when I have too many select time-outs when reading the card in C code and then changing to another channel and setting it back to the one I want. It works but it is just a workaround...

zimiao815, 2017-05-09:
thank you all，I according to your explanation，Modify the code，The original short time is deaf。Can now listen to 10 hours  more。
Have the effect ！！！
According to the following debug information I should do next？？？？ 
thank you very mach

device：AR9271USB
`

[321921.310000] ath: phy21: ath9k_htc_ani_work()----------debug for me
[321921.310000] ath: phy21: shortcal @32162131
[321921.310000] ath: phy21: listenTime=22232025 OFDM:1 errs=99/s CCK:5 errs=150/s ofdm_turn=1
[321921.410000] ath: phy21: ath9k_htc_ani_work()
[321921.410000] ath: phy21: listenTime=22232122 OFDM:1 errs=99/s CCK:5 errs=150/s ofdm_turn=1
[321921.510000] ath: phy21: ath9k_htc_ani_work()
[321921.510000] ath: phy21: listenTime=22232219 OFDM:1 errs=99/s CCK:5 errs=150/s ofdm_turn=1
[321921.610000] ath: phy21: ath9k_htc_ani_work()
[321921.610000] ath: phy21: listenTime=22232316 OFDM:1 errs=99/s CCK:5 errs=150/s ofdm_turn=1
[321921.710000] ath: phy21: ath9k_htc_ani_work()
[321921.710000] ath: phy21: listenTime=22232413 OFDM:1 errs=99/s CCK:5 errs=150/s ofdm_turn=1
root@MARKNUM:~# ifconfig mon21
mon21     Link encap:UNSPEC  HWaddr 00-12-7B-20-55-C8-7F-86-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:34265001 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8844607858 (8.2 GiB)  TX bytes:0 (0.0 B)

root@MARKNUM:~# ifconfig mon21
mon21     Link encap:UNSPEC  HWaddr 00-12-7B-20-55-C8-7F-BA-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:34265001 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8844607858 (8.2 GiB)  TX bytes:0 (0.0 B)
`

erikarn, 2017-05-09:
heh, send me a diff :)

-adrian

zimiao815, 2017-05-09:
`
 
Index:/drivers/net/wireless/ath/ath9k/init.c
===================================================================
---/drivers/net/wireless/ath/ath9k/init.c	2017-05-02 14:04:02.380706947 +0800
+++/drivers/net/wireless/ath/ath9k/init.c	2017-05-05 15:52:42.602669459 +0800

@@ -660,7 +660,9 @@
 		goto err_btcoex;
 
 	ath9k_cmn_init_crypto(sc->sc_ah);
+	printk(KERN_DEBUG "ath: ath9k_init_softc()---ath9k_init_misc()\n");
 	ath9k_init_misc(sc);
+	ath_check_ani(sc);
 	ath_fill_led_pin(sc);
 	ath_chanctx_init(sc);
 	ath9k_offchannel_init(sc);

 

/drivers/net/wireless/ath/ath9k/htc_drv_main.c
===================================================================
--- /drivers/net/wireless/ath/ath9k/htc_drv_main.c	2017-05-04 18:40:15.596674578 +0800
+++ /drivers/net/wireless/ath/ath9k/htc_drv_main.c	2017-05-05 19:04:01.286877126 +0800
@@ -421,7 +421,7 @@
 	priv->nstations++;
 	priv->vif_sta_pos[priv->mon_vif_idx] = sta_idx;
 	priv->ah->is_monitoring = true;
-
+	
 	ath_dbg(common, CONFIG,
 		"Attached a monitor interface at idx: %d, sta idx: %d\n",
 		priv->mon_vif_idx, sta_idx);
@@ -760,6 +760,7 @@
 	ieee80211_queue_delayed_work(common->hw, &priv->ani_work,
 				     msecs_to_jiffies(ATH_ANI_POLLINTERVAL));
 	
+	schedule_work(&priv->ani_work);
 	ath_dbg(common, ANI, "ath9k_htc_start_ani()\n");
 	
 }
@@ -1243,6 +1246,8 @@
 		ath9k_cmn_update_txpow(priv->ah, priv->curtxpow,
 				       priv->txpowlimit, &priv->curtxpow);
 	}
+	ath_dbg(common, ANI,"ath_htc: MONITOR is start \n");
+	ath9k_htc_start_ani(priv);
 
 out:
 	ath9k_htc_ps_restore(priv);
`

zimiao815, 2017-05-09:
but  a long time will still be in the future is deaf。should do ?

rodizio1, 2017-06-18:
I've tried this now, seems to work. That is, ANI seems to be enabled, I get periodic log messages when debug is enabled. Haven't had time yet to do any real-world testing though.

I've tried creating a patch using this commandline:
```
diff -u drivers/net/wireless/ath/ath9k/htc_drv_main.c drivers/net/wireless/ath/ath9k/htc_drv_main.c.ani 
--- drivers/net/wireless/ath/ath9k/htc_drv_main.c	2017-06-18 15:46:27.000000000 +0200
+++ drivers/net/wireless/ath/ath9k/htc_drv_main.c.ani	2017-06-18 15:58:22.763173760 +0200
@@ -771,6 +771,8 @@
 
 	ieee80211_queue_delayed_work(common->hw, &priv->ani_work,
 				     msecs_to_jiffies(ATH_ANI_POLLINTERVAL));
+	schedule_work(&priv->ani_work);
+	ath_dbg(common, ANI, "ath9k_htc_start_ani()\n");
 }
 
 void ath9k_htc_stop_ani(struct ath9k_htc_priv *priv)
@@ -1260,6 +1262,9 @@
 				       priv->txpowlimit, &priv->curtxpow);
 	}
 
+	ath_dbg(common, ANI,"ath_htc: MONITOR is start\n");
+	ath9k_htc_start_ani(priv);
+
 out:
 	ath9k_htc_ps_restore(priv);
 	mutex_unlock(&priv->mutex);
```

However, when I try to apply it, I get this:
```
patch -p0 < ani.patch 
patching file drivers/net/wireless/ath/ath9k/htc_drv_main.c
Hunk #1 FAILED at 771.
Hunk #2 succeeded at 1260 with fuzz 2.
1 out of 2 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/ath9k/htc_drv_main.c.rej

```
Sorry, don't have time to bother with this 1970ies unix software weirdness now, maybe somebody else can create a proper patch.