qca/open-ath9k-htc-firmware #129: make: *** No rule to make target 'install-linux'.  Stop.

quemexordium, 2017-05-04:
Hey. I'm trying to do this installation on Fedora 25 (4.10.13-200.fc25.x86_64 Kernel Version). I followed this guide for rookies: https://github.com/qca/open-ath9k-htc-firmware/pull/71/commits/a028f2fcdfa52b00ea0dc2d51c031c6c45c1d48e

Everything went perfectly until I ran this command: `sudo make install-linux`
The error was: `make: *** No rule to make target 'install-linux'.  Stop.`

It's probably because my operation system is not debian based but I don't know how to do it in my os.

olerem, 2017-05-04:
- this patch was never committed, so make file don't support an option "make install-linux"
- if your adapter is not seen in lsusb, then most probably something else is wrong. You don't need a firmware to see in in lsusb.
Take a look in the dmesg if there are some error messages. Probably some usb cable issue.

quemexordium, 2017-05-04:
What exactly should I look for in `dmesg`? I'm sorry if I'm being too much.

I've tried the same device on Windows and it worked perfectly fine. So it can't be some usb cable issue. In fact, it appears to be a kernel issue. I don't know what's wrong with Fedora but there are so many people on the net complaining about the same thing as me. I don't want to give up on Fedora that easily.

olerem, 2017-05-04:
compare demsg before you plug the usb adapter and after. This would be good starting point for troubleshooting.
I don't use fedora, so i can't say for sure what exactly is wrong.

quemexordium, 2017-05-04:
What's your current OS? I think I should change the OS rather than wasting my days.

olerem, 2017-05-04:
Currently i use Ubuntu 17.04.

quemexordium, 2017-05-04:
Okay I'm going to try openSUSE and if it doesn't work I'll involuntarily try Ubuntu. Let's see how it goes.

quemexordium, 2017-05-07:
I thought my usb wireless device had Atheros chipset but it turned out to have rtl8188eu. I'm closing the issue.

closed 2017-05-07 by quemexordium