qca/open-ath9k-htc-firmware #130: Tool chain build - gcc won't compile

sumeetkpatel, 2017-05-24:
Trying to build the toolchain on an OrangePi Lite using Armbian nightly.

Main compilation error:
```
checking for suffix of object files... configure: error: in `/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0/xtensa-elf/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
Makefile:10972: recipe for target 'configure-target-libgcc' failed
make[2]: *** [configure-target-libgcc] Error 1
make[2]: Leaving directory '/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0'
Makefile:856: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0'
Makefile:149: recipe for target '/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0/.built' failed
make: *** [/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0/.built] Error 2
```

/toolchain/build/gcc-6.2.0/xtensa-elf/libgcc/config.log [config.tar.gz](https://github.com/qca/open-ath9k-htc-firmware/files/1027414/config.tar.gz)

Currently installed GCC:
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4)

paulfertser, 2017-05-25:
Hello,

On Wed, May 24, 2017 at 03:39:33PM -0700, Sumeet Patel wrote:
> checking for suffix of object files... configure: error: in `/root/open-ath9k-htc-firmware-master/toolchain/build/gcc-6.2.0/xtensa-elf/libgcc':
> configure: error: cannot compute suffix of object files: cannot compile
> See `config.log' for more details.

So please attach full config.log.

paulfertser, 2017-05-25:
Eh, sorry, Github mail confused me as it didn't say anything explicit
about the attached files.

Are you following the procedure as described an README to the letter?
It probably looks like unpatched xtensa binutils are used.

olerem, 2017-05-25:
as far as i can see, we can't build currently it on armhf:
https://tests.reproducible-builds.org/debian/rb-pkg/unstable/armhf/open-ath9k-htc-firmware.html

paulfertser, 2017-05-26:
On Thu, May 25, 2017 at 04:01:30AM -0700, Oleksij Rempel wrote:
> as far as i can see, we can't build currently it on armhf:

Reproduced on my armhf laptop, and apparently it's the xtensa binutils
build itself that's failing straight away. Will try digging later...

paulfertser, 2017-05-27:
On Thu, May 25, 2017 at 04:01:30AM -0700, Oleksij Rempel wrote:
> as far as i can see, we can't build currently it on armhf:
> [1]https://tests.reproducible-builds.org/debian/rb-pkg/unstable/armhf/open-ath9k-htc-firmware.html

I suggest those affected by this build failure to test adding this
patch to gcc build:

https://paulfertser.info/gcc-xtensa-fix-entry.patch

jcmvbkbc, 2017-05-30:
I believe the following patch is the right solution to this issue: https://github.com/jcmvbkbc/gcc-xtensa/commit/f89616b932b9460ebff3167e363f1d6ac2fa1445.patch

paulfertser, 2017-05-30:
On Mon, May 29, 2017 at 08:45:49PM -0700, Max Filippov wrote:
> I believe the following patch is the right solution to this issue:
> [1]https://github.com/jcmvbkbc/gcc-xtensa/commit/f89616b932b9460ebff3167e363f1d6ac2fa1445.patch

Thank you so much Max! It is the right solution indeed, I confirm it's
working as intended. And I've actually noticed the gcc warnings
regarding incorrect format strings when building the unpatched
xtensa.c but ignored those, silly me.

I guess it would probably make sense to have -Werror- for those, as
codegeneration depends on fprintf working correctly in this case.

As a sidenote, for those building with non-HEAD gcc version, the
ChangeLog part of the patch needs to be stripped, but other than that,
it applies nicely to 6.3.0.