qca/open-ath9k-htc-firmware #134: Maximum MTU?

rodizio1, 2017-08-07:
I've recently tried injecting frames in monitor mode with more than 1500bytes with Ar9271 cards on Kernel 4.9.35. Set interface MTU to 2300 with ifconfig. Anything higher than 1560 bytes payload (not sure right now how much that is with 802.11 header and fcs, I think something around 1600 or so, but I can look it up exactly if needed) makes something 'freeze'. From that point on, injecting frames is not possible anymore at all, also not with smaller sizes. Solved it by rebooting.

Doing the same with Ralink rt2800usb cards works up to 2270 bytes payload, so I assume it's not userspace, the kernel or it's network/mac stack, but something in the ath9k kernel drivers or ath9k firmware (?)

Have grepped the ath9k kernel drivers and firmware for 'mtu' quickly, couldn't find anything.

Does anybody know where to look or can give some hints?

erikarn, 2017-08-07:
hi,

I think the firmware needs to be tweaked to allow larger frames, which
means it'll be able to buffer less total numbers of frames.

-adrian

olerem, 2017-08-08:
Can you please test if reducing size of MAX_TX_BUF_SIZE will prevent freezing. If yes, this will probably right field for research.

olerem, 2017-08-08:
... on the drivers side. see hif_usb.*
this controller has at least one known usb packet size related issue: https://github.com/qca/open-ath9k-htc-firmware/wiki/usb-related-issues (EP4 issue)
I won't be surprised if there can be more.

rodizio1, 2017-08-08:
Thanks a lot, I'll give that a try (and also test on an Intel x86 machine to make sure it's not some strange Raspberry USB Problem).

rodizio1, 2017-08-12:
Gave it a quick test. 16384 makes the Pi crash completely, 28000 makes no difference as far as I can see.

olerem, 2017-08-12:
What is the meaning of this numbers?

rodizio1, 2017-08-13:
The MAX_TX_BUF_SIZE in hif_usb.h.

rodrigowue, 2020-12-04:
Hey @rodizio1 ,

Do you have any updates on this? 

Cheers,
Rodrigo