qca/open-ath9k-htc-firmware #136: Debian stretch: Mac Chip Rev 0x0f.3 is not supported by this driver

n055, 2017-09-16:
I'm trying to use a AR9271 dongle on my ASUS C201 running native Debian stretch (armhf, mainline kernel). The wifi dongle gets partway through initializing, but after the firmware is copied over, it gives up.

dmesg:

```
[ 1201.550915] usb 3-1: new high-speed USB device number 8 using dwc2
[ 1201.786290] usb 3-1: New USB device found, idVendor=0cf3, idProduct=9271
[ 1201.793050] usb 3-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 1201.800511] usb 3-1: Product: UB93
[ 1201.803957] usb 3-1: Manufacturer: ATHEROS
[ 1201.808094] usb 3-1: SerialNumber: 12345
[ 1201.812515] usb 3-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.dev.0.fw requested
[ 1201.820459] usb 3-1: firmware: direct-loading firmware ath9k_htc/htc_9271-1.dev.0.fw
[ 1202.105406] usb 3-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.dev.0.fw, size: 51008
[ 1202.357096] ath9k_htc 3-1:1.0: ath9k_htc: HTC initialized with 33 credits
[ 1202.890913] ath: phy5: Mac Chip Rev 0x0f.3 is not supported by this driver
[ 1202.897852] ath: phy5: Unable to initialize hardware; initialization status: -95
[ 1202.905311] ath: phy5: Unable to initialize hardware; initialization status: -95
[ 1202.912780] ath9k_htc: Failed to initialize the device
[ 1202.918653] usb 3-1: ath9k_htc: USB layer deinitialized
```

olerem, 2017-09-16:
this is strange: "ath: phy5: Mac Chip Rev 0x0f.3 is not supported by this drive"
Have you tried to use FW out of debian repository? For example this one https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/ath9k_htc

n055, 2017-09-17:
I downloaded the firmware in a .deb from https://packages.debian.org/buster/firmware-ath9k-htc and used apt to install, since I'm using Debian stable and the firmware is only in testing/unstable.

I tried installing the firmware you linked to, and it gives me the same error.

However, I discovered that if I connect the dongle before turning on my C201 laptop, I don't get the error. I also fixed some issues preventing me from bringing up the wlan0 interface based on steps I found at https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=839605:

```
# touch /etc/systemd/network/99-default.link
# cat > /etc/NetworkManager/NetworkManager.conf <<EOF
[device-mac-randomization]
wifi.scan-rand-mac-address=no
EOF
```

The "Mac Chip Rev 0x0f.3 is not supported by this driver" error still appears if, after the laptop boots, I disconnect and then reconnect the wifi dongle, but as long as it's connected before boot, it seems to work.

olerem, 2017-09-17:
hm after googling for "Mac Chip Rev 0x0f.3 is not supported by this driver", i found more reports with this issue. even here in this tracker. All seems to be related to ARM platform, may be to some usb controller or driver.
Not sure if I can debug this issue remotely. Good start will be to make a capture of usb traffic with wireshark.

n055, 2017-11-05:
So, 1.5 months later, I had some time to take a look at this problem again, and found a "solution" besides booting with the adapter already connected. Hotplugging he adapter seems to work fine if it's plugged into a USB hub. Like you said, must be some weirdness with the USB on this ARM platform.

closed 2017-11-05 by n055