qca/open-ath9k-htc-firmware #137: AR9271 random freezes in AP mode when distant client is connected

rlking, 2017-09-20:
I have a RPI 3 with raspbian stretch using a TL-WN722N in AP mode with 802.11n enabled.
It runs stable for days until I start connecting to it from a distant room, then I get unpredictable freezes
within hours and only physically detaching the stick fixes the issue. I tried most of the 4.9.x kernels
and also 4.12.x but it doesn't make any differene. I'm not sure if this is kernel/driver/firmware related.

Here is some output

[Mit Sep 20 16:01:35 2017] usb 1-1.4: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[Mit Sep 20 16:01:35 2017] usbcore: registered new interface driver ath9k_htc
[Mit Sep 20 16:01:35 2017] usb 1-1.4: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[Mit Sep 20 16:01:35 2017] ath9k_htc 1-1.4:1.0: ath9k_htc: HTC initialized with 33 credits
[Mit Sep 20 16:01:36 2017] ath9k_htc 1-1.4:1.0: ath9k_htc: FW Version: 1.4
[Mit Sep 20 16:01:36 2017] ath9k_htc 1-1.4:1.0: FW RMW support: On
[Mit Sep 20 16:01:36 2017] ath: EEPROM regdomain: 0x809c
[Mit Sep 20 16:01:36 2017] ath: EEPROM indicates we should expect a country code
[Mit Sep 20 16:01:36 2017] ath: doing EEPROM country->regdmn map search
[Mit Sep 20 16:01:36 2017] ath: country maps to regdmn code: 0x52
[Mit Sep 20 16:01:36 2017] ath: Country alpha2 being used: CN
[Mit Sep 20 16:01:36 2017] ath: Regpair used: 0x52
[Mit Sep 20 16:01:36 2017] ieee80211 phy3: Atheros AR9271 Rev:1
[Mit Sep 20 16:01:36 2017] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready

Sep 17 22:01:56 kingcastle kernel: [41654.576377] INFO: task kworker/2:1:9370 blocked for more than 120 seconds.
Sep 17 22:01:56 kingcastle kernel: [41654.576388]       Not tainted 4.9.50-v7+ #1035
Sep 17 22:01:56 kingcastle kernel: [41654.576393] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 17 22:01:56 kingcastle kernel: [41654.576399] kworker/2:1     D    0  9370      2 0x00000000
Sep 17 22:01:56 kingcastle kernel: [41654.576425] Workqueue: events linkwatch_event
Sep 17 22:01:56 kingcastle kernel: [41654.576463] [<80716230>] (__schedule) from [<80716754>] (schedule+0x50/0xa8)
Sep 17 22:01:56 kingcastle kernel: [41654.576482] [<80716754>] (schedule) from [<80716b48>] (schedule_preempt_disabled+0x18/0x1c)
Sep 17 22:01:56 kingcastle kernel: [41654.576498] [<80716b48>] (schedule_preempt_disabled) from [<80718264>] (__mutex_lock_slowpath+0xac/0x164)
Sep 17 22:01:56 kingcastle kernel: [41654.576511] [<80718264>] (__mutex_lock_slowpath) from [<8071837c>] (mutex_lock+0x60/0x64)
Sep 17 22:01:56 kingcastle kernel: [41654.576523] [<8071837c>] (mutex_lock) from [<8063c4f0>] (rtnl_lock+0x20/0x24)
Sep 17 22:01:56 kingcastle kernel: [41654.576539] [<8063c4f0>] (rtnl_lock) from [<80642a28>] (linkwatch_event+0x18/0x44)
Sep 17 22:01:56 kingcastle kernel: [41654.576557] [<80642a28>] (linkwatch_event) from [<80136720>] (process_one_work+0x158/0x454)
Sep 17 22:01:56 kingcastle kernel: [41654.576580] [<80136720>] (process_one_work) from [<80136a80>] (worker_thread+0x64/0x568)
Sep 17 22:01:56 kingcastle kernel: [41654.576595] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
Sep 17 22:01:56 kingcastle kernel: [41654.576612] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Sep 17 22:01:56 kingcastle kernel: [41654.576623] INFO: task kworker/3:2:9683 blocked for more than 120 seconds.
Sep 17 22:01:56 kingcastle kernel: [41654.576628]       Not tainted 4.9.50-v7+ #1035
Sep 17 22:01:56 kingcastle kernel: [41654.576632] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 17 22:01:56 kingcastle kernel: [41654.576636] kworker/3:2     D    0  9683      2 0x00000000
Sep 17 22:01:56 kingcastle kernel: [41654.576880] Workqueue: ipv6_addrconf addrconf_verify_work [ipv6]
Sep 17 22:01:56 kingcastle kernel: [41654.576903] [<80716230>] (__schedule) from [<80716754>] (schedule+0x50/0xa8)
Sep 17 22:01:56 kingcastle kernel: [41654.576920] [<80716754>] (schedule) from [<80716b48>] (schedule_preempt_disabled+0x18/0x1c)
Sep 17 22:01:56 kingcastle kernel: [41654.576936] [<80716b48>] (schedule_preempt_disabled) from [<80718264>] (__mutex_lock_slowpath+0xac/0x164)
Sep 17 22:01:56 kingcastle kernel: [41654.576952] [<80718264>] (__mutex_lock_slowpath) from [<8071837c>] (mutex_lock+0x60/0x64)
Sep 17 22:01:56 kingcastle kernel: [41654.576965] [<8071837c>] (mutex_lock) from [<8063c4f0>] (rtnl_lock+0x20/0x24)
Sep 17 22:01:56 kingcastle kernel: [41654.577183] [<8063c4f0>] (rtnl_lock) from [<7f00dbf8>] (addrconf_verify_work+0x18/0x24 [ipv6])
Sep 17 22:01:56 kingcastle kernel: [41654.577408] [<7f00dbf8>] (addrconf_verify_work [ipv6]) from [<80136720>] (process_one_work+0x158/0x454)
Sep 17 22:01:56 kingcastle kernel: [41654.577426] [<80136720>] (process_one_work) from [<80136a80>] (worker_thread+0x64/0x568)
Sep 17 22:01:56 kingcastle kernel: [41654.577442] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
Sep 17 22:01:56 kingcastle kernel: [41654.577461] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Sep 17 22:03:59 kingcastle kernel: [41777.457610] INFO: task kworker/2:1:9370 blocked for more than 120 seconds.
Sep 17 22:03:59 kingcastle kernel: [41777.457621]       Not tainted 4.9.50-v7+ #1035
Sep 17 22:03:59 kingcastle kernel: [41777.457625] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 17 22:03:59 kingcastle kernel: [41777.457631] kworker/2:1     D    0  9370      2 0x00000000
Sep 17 22:03:59 kingcastle kernel: [41777.457657] Workqueue: events linkwatch_event
Sep 17 22:03:59 kingcastle kernel: [41777.457690] [<80716230>] (__schedule) from [<80716754>] (schedule+0x50/0xa8)
Sep 17 22:03:59 kingcastle kernel: [41777.457712] [<80716754>] (schedule) from [<80716b48>] (schedule_preempt_disabled+0x18/0x1c)
Sep 17 22:03:59 kingcastle kernel: [41777.457729] [<80716b48>] (schedule_preempt_disabled) from [<80718264>] (__mutex_lock_slowpath+0xac/0x164)
Sep 17 22:03:59 kingcastle kernel: [41777.457742] [<80718264>] (__mutex_lock_slowpath) from [<8071837c>] (mutex_lock+0x60/0x64)
Sep 17 22:03:59 kingcastle kernel: [41777.457754] [<8071837c>] (mutex_lock) from [<8063c4f0>] (rtnl_lock+0x20/0x24)
Sep 17 22:03:59 kingcastle kernel: [41777.457770] [<8063c4f0>] (rtnl_lock) from [<80642a28>] (linkwatch_event+0x18/0x44)
Sep 17 22:03:59 kingcastle kernel: [41777.457788] [<80642a28>] (linkwatch_event) from [<80136720>] (process_one_work+0x158/0x454)
Sep 17 22:03:59 kingcastle kernel: [41777.457807] [<80136720>] (process_one_work) from [<80136a80>] (worker_thread+0x64/0x568)
Sep 17 22:03:59 kingcastle kernel: [41777.457827] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
Sep 17 22:03:59 kingcastle kernel: [41777.457845] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Sep 17 22:03:59 kingcastle kernel: [41777.457856] INFO: task kworker/3:2:9683 blocked for more than 120 seconds.
Sep 17 22:03:59 kingcastle kernel: [41777.457861]       Not tainted 4.9.50-v7+ #1035
Sep 17 22:03:59 kingcastle kernel: [41777.457865] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[... 69 lines cut ...]
Sep 17 22:32:11 kingcastle kernel: [43469.255626] [<8011d348>] (__warn) from [<8011d3a8>] (warn_slowpath_fmt+0x48/0x50)
Sep 17 22:32:11 kingcastle kernel: [43469.255987] [<8011d3a8>] (warn_slowpath_fmt) from [<7f3b6620>] (ieee80211_free_ack_frame+0x5c/0x60 [mac80211])
Sep 17 22:32:11 kingcastle kernel: [43469.256253] [<7f3b6620>] (ieee80211_free_ack_frame [mac80211]) from [<80456dac>] (idr_for_each+0xb0/0xe4)
Sep 17 22:32:11 kingcastle kernel: [43469.256518] [<80456dac>] (idr_for_each) from [<7f3b65a0>] (ieee80211_free_hw+0x4c/0x70 [mac80211])
Sep 17 22:32:11 kingcastle kernel: [43469.256775] [<7f3b65a0>] (ieee80211_free_hw [mac80211]) from [<7f584454>] (ath9k_htc_disconnect_device+0x9c/0xa0 [ath9k_htc])
Sep 17 22:32:11 kingcastle kernel: [43469.256858] [<7f584454>] (ath9k_htc_disconnect_device [ath9k_htc]) from [<7f57b8e8>] (ath9k_htc_hw_deinit+0x20/0x24 [ath9k_htc])
Sep 17 22:32:11 kingcastle kernel: [43469.256938] [<7f57b8e8>] (ath9k_htc_hw_deinit [ath9k_htc]) from [<7f57c778>] (ath9k_hif_usb_disconnect+0x104/0x12c [ath9k_htc])
Sep 17 22:32:11 kingcastle kernel: [43469.256996] [<7f57c778>] (ath9k_hif_usb_disconnect [ath9k_htc]) from [<80560578>] (usb_unbind_interface+0x8c/0x2a0)
Sep 17 22:32:11 kingcastle kernel: [43469.257033] [<80560578>] (usb_unbind_interface) from [<804fc718>] (__device_release_driver+0x94/0x118)
[... cut ...]
Sep 17 22:32:11 kingcastle kernel: [43469.257303] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
Sep 17 22:32:11 kingcastle kernel: [43469.257340] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Sep 17 22:32:11 kingcastle kernel: [43469.257353] ---[ end trace fd8740d213110681 ]---
Sep 17 22:32:11 kingcastle kernel: [43469.257523] usb 1-1.4: ath9k_htc: USB layer deinitialized

If you have any suggestions how to further debug this just tell me.

Kind regards

olerem, 2017-09-20:
It would be really interesting to get log from firmware. Can you solder and have some experience with UART?
Here you can find info about this chip https://wikidevi.com/wiki/Atheros_AR9271

rlking, 2017-09-20:
I will look into this, I should have everything

rlking, 2017-09-20:
I bricked it while removing the internal antenna 😔

olerem, 2017-09-21:
In case you still wont to play with this chip, i would recommend this model https://wikidevi.com/wiki/ALFA_Network_AWUS036NHA
it has accessible uart pins and external antenna.

olerem, 2017-09-27:
For RPi users, if you find a way to power cycle the usb port, there should be no problem to reset the usb adapter. Some thing like this https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=93463#p651809

dagf2101, 2018-07-06:
I have a really similar issue in Ad-Hoc mode on kernel 4.15. We are using a network of around 8 devices connected to NUC PCs in our tests. Every time a device disconnect, other devices have a chance to crash. I reproduce it by rebooting one of the PC and eventually other devices start crashing the kernel thread freeze on those crashed nodes.

Here is the kernel log:

Jun 20 14:20:23 BTDN7440035E kernel: INFO: task kworker/0:1:34 blocked for more than 120 seconds.
Jun 20 14:20:23 BTDN7440035E kernel:       Tainted: G        W    L   4.15.0-kali2-amd64 #1 Debian 4.15.11-1kali1
Jun 20 14:20:23 BTDN7440035E kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 20 14:20:23 BTDN7440035E kernel: kworker/0:1     D    0    34      2 0x80000000
Jun 20 14:20:23 BTDN7440035E kernel: Workqueue: ipv6_addrconf addrconf_verify_work
Jun 20 14:20:23 BTDN7440035E kernel: Call Trace:
Jun 20 14:20:23 BTDN7440035E kernel:  ? __schedule+0x297/0x8b0
Jun 20 14:20:23 BTDN7440035E kernel:  schedule+0x28/0x80
Jun 20 14:20:23 BTDN7440035E kernel:  schedule_preempt_disabled+0xa/0x10
Jun 20 14:20:23 BTDN7440035E kernel:  __mutex_lock.isra.1+0x1a0/0x4e0
Jun 20 14:20:23 BTDN7440035E kernel:  ? __switch_to_asm+0x40/0x70
Jun 20 14:20:23 BTDN7440035E kernel:  ? addrconf_verify_work+0xa/0x20
Jun 20 14:20:23 BTDN7440035E kernel:  addrconf_verify_work+0xa/0x20
Jun 20 14:20:23 BTDN7440035E kernel:  process_one_work+0x17b/0x370
Jun 20 14:20:23 BTDN7440035E kernel:  worker_thread+0x2e/0x390
Jun 20 14:20:23 BTDN7440035E kernel:  ? process_one_work+0x370/0x370
Jun 20 14:20:23 BTDN7440035E kernel:  kthread+0x113/0x130
Jun 20 14:20:23 BTDN7440035E kernel:  ? kthread_create_worker_on_cpu+0x70/0x70
Jun 20 14:20:23 BTDN7440035E kernel:  ret_from_fork+0x35/0x40
Jun 20 14:20:30 BTDN7440035E kernel: watchdog: BUG: soft lockup - CPU#2 stuck for 22s! [wpa_supplicant:1184]
Jun 20 14:20:30 BTDN7440035E kernel: Modules linked in: xt_mac iptable_filter hid_generic usbhid hid tun ctr ccm fuse snd_hda_codec_hdmi snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_soc_acpi snd_soc_core snd_compress binfmt_misc arc4 i2c_designware_platform i2c_designware_core wmi_bmof evdev intel_rapl x86_pkg_temp_thermal intel_powerclamp kvm_intel btusb btrtl btbcm kvm btintel irqbypass iwlmvm bluetooth intel_cstate ath9k_htc ath9k_common snd_hda_intel drbg intel_uncore ath9k_hw snd_hda_codec snd_hda_core ansi_cprng ath snd_hwdep idma64 snd_pcm intel_rapl_perf iwlwifi snd_timer mac80211 i915 intel_lpss_pci pcspkr snd ecdh_generic iTCO_wdt cfg80211 drm_kms_helper rfkill drm mei_me soundcore iTCO_vendor_support sg intel_lpss mei i2c_algo_bit intel_pch_thermal shpchp wmi mfd_core video
Jun 20 14:20:30 BTDN7440035E kernel:  tpm_crb acpi_pad button coretemp ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic fscrypto ecb sd_mod crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd ahci libahci e1000e ptp xhci_pci pps_core libata xhci_hcd i2c_i801 scsi_mod usbcore usb_common fan thermal
Jun 20 14:20:30 BTDN7440035E kernel: CPU: 2 PID: 1184 Comm: wpa_supplicant Tainted: G        W    L   4.15.0-kali2-amd64 #1 Debian 4.15.11-1kali1
Jun 20 14:20:30 BTDN7440035E kernel: Hardware name: Intel Corporation NUC7i5DNB/NUC7i5DNB, BIOS DNKBLi5v.86A.0040.2018.0315.1451 03/15/2018
Jun 20 14:20:30 BTDN7440035E kernel: RIP: 0010:try_to_grab_pending+0xfb/0x150
Jun 20 14:20:30 BTDN7440035E kernel: RSP: 0018:ffffb11703f9b8a8 EFLAGS: 00000286 ORIG_RAX: ffffffffffffff11
Jun 20 14:20:30 BTDN7440035E kernel: RAX: 0000000000000000 RBX: ffff91ca5a365928 RCX: 0000000000000000
Jun 20 14:20:30 BTDN7440035E kernel: RDX: 0000000000000001 RSI: 0000000000000000 RDI: 0000000000000286
Jun 20 14:20:30 BTDN7440035E kernel: RBP: ffffb11703f9b8d0 R08: ffff91ca60000b68 R09: ffff91ca70c210c0
Jun 20 14:20:30 BTDN7440035E kernel: R10: 0000000000000000 R11: 0000000000000040 R12: ffff91ca70c210c0
Jun 20 14:20:30 BTDN7440035E kernel: R13: ffffb11703f9b8f0 R14: ffffffffbae8e800 R15: ffff91ca565cccb0
Jun 20 14:20:30 BTDN7440035E kernel: FS:  00007f983bfe4800(0000) GS:ffff91ca70d00000(0000) knlGS:0000000000000000
Jun 20 14:20:30 BTDN7440035E kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 20 14:20:30 BTDN7440035E kernel: CR2: 0000558b5274e818 CR3: 0000000421a2c002 CR4: 00000000003606e0
Jun 20 14:20:30 BTDN7440035E kernel: DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Jun 20 14:20:30 BTDN7440035E kernel: DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Jun 20 14:20:30 BTDN7440035E kernel: Call Trace:
Jun 20 14:20:30 BTDN7440035E kernel:  __cancel_work_timer+0x42/0x1b0
Jun 20 14:20:30 BTDN7440035E kernel:  ath9k_htc_sta_remove+0x21/0x80 [ath9k_htc]
Jun 20 14:20:30 BTDN7440035E kernel:  drv_sta_state+0x25b/0x3f0 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  sta_info_move_state+0x181/0x260 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  __sta_info_destroy_part2+0x54/0x110 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  __sta_info_flush+0x150/0x1b0 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  ieee80211_ibss_disconnect+0x4b/0x2e0 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  ieee80211_ibss_leave+0x16/0xb0 [mac80211]
Jun 20 14:20:30 BTDN7440035E kernel:  __cfg80211_leave_ibss+0x4b/0x100 [cfg80211]
Jun 20 14:20:30 BTDN7440035E kernel:  cfg80211_leave_ibss+0x35/0x50 [cfg80211]
[... cut ...]
Jun 20 14:20:30 BTDN7440035E kernel: RBP: 0000556cbf896880 R08: 0000000000000000 R09: 000000000000000d
Jun 20 14:20:30 BTDN7440035E kernel: R10: 0000556cbf893010 R11: 0000000000000246 R12: 0000556cbf8cae10
Jun 20 14:20:30 BTDN7440035E kernel: R13: 00007ffe99a04a90 R14: 0000000000000000 R15: 0000000000000000
Jun 20 14:20:30 BTDN7440035E kernel: Code: 48 83 c8 01 4c 89 e7 48 89 03 c6 07 00 0f 1f 40 00 b8 01 00 00 00 e9 4d ff ff ff 4c 89 e7 c6 07 00 0f 1f 40 00 48 8b 7d 00 57 9d <0f> 1f 44 00 00 48 8b 13 b8 fe ff ff ff 83 e2 14 48 83 fa 10 0f