qca/open-ath9k-htc-firmware #138: Makefile for Android

FFY00, 2017-09-27:
Can you add android support to the Makefile? I want to build the driver modules for Android (aarch64) but I don't have a lot of time to edit every Makefile/CMakeLists. I've tweaked a bit the makefiles, and the toolchain directory structure, to allow me to build it. The current error I get is about `cc1` which is probably not included in the toolchain or it's in a different directory.
```error trying to exec 'cc1': execvp: No such file or directory```
If you want to add full support for android I recommend to use [UberTC](https://bitbucket.org/UBERTC/).

olerem, 2017-09-28:
patches are welcome :)

FFY00, 2017-09-28:
Ok, if I have time I will make a patch.

FFY00, 2017-09-28:
Can someone tell me if `TARGET` is defined in any file other than the `Makefile` in the root directory? I edit it but it keeps `xtensa-elf`.

olerem, 2017-09-29:
~: git grep TARGET | grep elf
.....
Makefile:TARGET=xtensa-elf
target_firmware/configure:TARGET=xtensa-elf

FFY00, 2017-09-29:
What version of gcc is supposed to be used when building this? I tried with gcc 5.4 (linaro) but I got an error.
```
error: conflicting types for ‘ath_hal_malloc'
```

olerem, 2017-09-29:
grep _VER Makefile
GMP_VER=6.1.1
GMP_URL=https://ftp.gnu.org/gnu/gmp/gmp-$(GMP_VER).tar.bz2
GMP_TAR=gmp-$(GMP_VER).tar.bz2
GMP_DIR=gmp-$(GMP_VER)
MPFR_VER=3.1.4
MPFR_URL=https://ftp.gnu.org/gnu/mpfr/mpfr-$(MPFR_VER).tar.bz2
MPFR_TAR=mpfr-$(MPFR_VER).tar.bz2
MPFR_DIR=mpfr-$(MPFR_VER)
MPC_VER=1.0.3
MPC_URL=https://ftp.gnu.org/gnu/mpc/mpc-$(MPC_VER).tar.gz
MPC_TAR=mpc-$(MPC_VER).tar.gz
MPC_DIR=mpc-$(MPC_VER)
BINUTILS_VER=2.27
BINUTILS_URL=https://ftp.gnu.org/gnu/binutils/binutils-$(BINUTILS_VER).tar.bz2
BINUTILS_TAR=binutils-$(BINUTILS_VER).tar.bz2
BINUTILS_DIR=binutils-$(BINUTILS_VER)
GCC_VER=6.3.0
GCC_URL=https://ftp.gnu.org/gnu/gcc/gcc-$(GCC_VER)/gcc-$(GCC_VER).tar.bz2
GCC_TAR=gcc-$(GCC_VER).tar.bz2
GCC_DIR=gcc-$(GCC_VER)

olerem, 2017-09-29:
beside, this firmware is only for xtensa, only for two chips ar9271 and ar7010. Why do you wont to compile it for aarch64?
Or do you wont to compile the toolchain fir aarch64?

FFY00, 2017-09-29:
I want to compile it for aarch64. I think the incompatible code is just for controlling the led.

oliv3r, 2018-01-11:
Hey, just a curious bystander, but why would you like to compile it for aarch64?
The firmware (which this repo is about) is a bin file that gets uploaded into the ar9271 (or the ar7010). So that will always be the xtensa 32 bit flavor and works no matter what host OS is being used (or host architecture for that matter), right?

So the only reason to add this support is to be able to _compile_ the firmware on aarch64, which is a valid usecase if you have a chromebook or whatever to compile on, but it's probably just as easy to spin up a virtual machine or a remote server just to compile it all. Or am I overlooking a usercase?

closed 2018-04-21 by FFY00