qca/open-ath9k-htc-firmware #140: Can I make the "interrupt function" to be called before receiving entire frame?

HyunjoongLee, 2018-03-06:
I want to make an AP (using hostapd) to transmit a buffered frame when the AP receives a beacon frame from other APs. (More specifically, during the beacon transmission periods, not after receiving)
Now, I am trying to implement the functionality in the "ath_intr" in firmware source code. 
But when I checked the receiving buffer memory in "ath_intr" function, it seems that entire frame is already received in the receiving buffer. Can I make the interrupt function to be called right after the AP starts receiving a frame, i.e., after receiving PHY or MAC header? 
or 
Which function calls "ath_intr"? Is it implemented in hardware ROM? I can't find where "ath_intr" is called. Any comments or advice will be helpful to me and I will appreciate for your help.
Thank you very much!

olerem, 2018-03-06:
target_firmware/magpie_fw_dev/target/init/init.c
wlan_task
  wlan_pci_isr
    g_wlan_intr
       adf_os_setup_intr
          ath_intr

HyunjoongLee, 2018-03-09:
@olerem Thank you very much. It helped me to understand how the function (ath_intr) is called. 

For others, 
In wlan_task(), there is a infinite loop, which calls wlan_pci_isr() periodically (maybe microseconds level).
Then, ath_intr() is called, and depending on the type of interrupts, each tasklet function is called.

HyunjoongLee, 2018-03-09:
dfdf

HyunjoongLee, 2018-03-09:
dfd

erikarn, 2018-03-09:
Hi!

So, no - there's no interrupt that happens when you START getting a frame.
Are you trying to do TDMA? There are other ways.

-adrian

HyunjoongLee, 2018-03-09:
@erikarn Hi Adrian, 
What do you mean with 'TDMA'? I am trying to make an AP to utilize air-time for its data transmission during adjacent AP's beacon transmission periods. Currently, I am working on making an AP to monitor receive buffer to check whether the received frame is transmitted from target AP. 

Hyunjoong,

erikarn, 2018-03-09:
Hi!

Ok. Well, you don't get an interrupt until the hardware has completed the
DMA of the frame you wish to receive. Sorry!

So the best you can do right now is trigger a transmit of a frame whenever
you receive something, and yeah that's done via something in ath_intr().

-a

makesens86, 2018-03-12:
@erikarn 
Thanks for your comments. 

By the way, I found the way to access receive buffer before the hardware complete the DMA of the current frame. As ath_intr() function is periodically called, and if there is no pending interrupt, it does nothing. But at that point,  I could still access the receive buffer memory to monitor it. Since the ath_intr() is called every few microseconds, i think this method can be used to check the receive buffer (almost) in real-time.

erikarn, 2018-03-12:
Hi,

Yes - the hardware starts flushing the frame out of the RX FIFO via DMA
shortly after it's received enough of the frame to know what to do, and
there's a config parameter marking the threshold at which to start this.
But you don't get any interrupt notification that it's started - you'd have
to monitor debug GPIO lines to detect the start of the frame and then poll
that buffer.

Definitely not a supported usecase!

-adrian