qca/open-ath9k-htc-firmware #141: Replacing buffered frame with a new frame (with different length and MCS)

makesens86, 2018-03-29:
Hello?

When an AP is waiting to access channel before its transmission, can a buffered frame be replaced with a new frame with different length and MCS? 

For example, let's assume there are 2 A-MPDU frames in the TX buffer, (one is consisting of subframes #1-5, and the other is consisting of subframes 6-10). In this case, is it possible to make new A-MPDU frame with subframe number 1-8? 

Right now, I am considering this way. 
First, flush out all the buffered A-MPDU frames and move all the subframes in the software queue and make a new A-MPDU frame with target length. 
Is there any other option to recommend? 

Thank you very much.

erikarn, 2018-03-29:
hi,

once it in the hw tx queue it needs to go out or be aborted and retried.

-a

makesens86, 2018-03-29:
@erikarn Hi, always appreciate for your comments. 
If it is aborted, is the original frame (multiple subframes) buffered software queue?
I will look into the src code about abort and retry. 

hyunjoong