qca/open-ath9k-htc-firmware #142: usb 2-1: firmware: failed to load ath9k_htc/htc_9271-1.4.0.fw (-2)

chefarov, 2018-04-14:
Thank you for providing these drivers.

I am using AWUS036NHA (Atheros 9271L). I have followed the build procedure outlined in readme, and I have moved the firmware to `/lib/firmware/htc_9271.fw`. What troubles me is I am getting these logs:

```
:~$ sudo dmesg -T | grep -i ath9k
[Sat Apr 14 21:36:50 2018] usb 2-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[Sat Apr 14 21:36:50 2018] usbcore: registered new interface driver ath9k_htc
[Sat Apr 14 21:36:50 2018] usb 2-1: firmware: failed to load ath9k_htc/htc_9271-1.4.0.fw (-2)
[Sat Apr 14 21:36:50 2018] usb 2-1: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2
[Sat Apr 14 21:36:50 2018] usb 2-1: ath9k_htc: Firmware htc_9271.fw requested
[Sat Apr 14 21:36:50 2018] usb 2-1: ath9k_htc: Transferred FW: htc_9271.fw, size: 51008
[Sat Apr 14 21:36:50 2018] ath9k_htc 2-1:1.0: ath9k_htc: HTC initialized with 33 credits
[Sat Apr 14 21:36:51 2018] ath9k_htc 2-1:1.0: ath9k_htc: FW Version: 1.4
[Sat Apr 14 21:36:51 2018] ath9k_htc 2-1:1.0: FW RMW support: On
```

It seems that it needs the specific version (htc_9271-1.4.0)? How do I view my current version?

I am using `debian sid (4.15.0-2-amd64)`.

PS: I am asking since I am experiencing some functionality issues and I am trying to debug the situation.

olerem, 2018-04-14:
On debian sid it should be enough to install firmware-ath9k-htc. This package is based on latest git version of this firmware.
https://tracker.debian.org/pkg/open-ath9k-htc-firmware

chefarov, 2018-04-16:
Installing the above package brought 
`/lib/firmware/ath9k_htc/htc_9271-1.dev.0.fw`. Now I am **not** getting any more of those logs:

```
[Sat Apr 14 21:36:50 2018] usb 2-1: firmware: failed to load ath9k_htc/htc_9271-1.4.0.fw (-2)
[Sat Apr 14 21:36:50 2018] usb 2-1: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2
```

Thank you!

closed 2018-04-16 by chefarov

saintofinternet, 2019-01-09:
> Installing the above package brought `/lib/firmware/ath9k_htc/htc_9271-1.dev.0.fw`.
>
> Now I am **not** getting any more of those logs:
>
> ```
> [Sat Apr 14 21:36:50 2018] usb 2-1: firmware: failed to [...]

how did you install that package?

chefarov, 2019-01-10:
It's has been a long time, but I think the debian package is `firmware-atheros`, thus via `apt install firmware-atheros`. I now see that there is also `firmware-ath9k-htc`, I suppose the former includes the later (?).

After that I had random continuous disconnection issues, which were solved after I uninstalled `wicd`.