qca/open-ath9k-htc-firmware #144: make error on target_firmware

SunTzuShin3, 2018-06-04:
Hello,

When I run # make -C target_firmware   i receive this error

[ -d build ] || ./configure || { rm -rf build; false; }
./configure: 55: ./configure: cmake: not found
Makefile:38: recipe for target 'all' failed
make: *** [all] Error 1

i have cmake installed in the home directory. some forums that have the make error suggest that cmake is not in the path. i'm very new to kali linux and don't know where to start with where the error is actually occurring. any help or guidance would be much appreciated.

olerem, 2018-06-04:
why do you need to compile firmware at all? it is already available as binary.

SunTzuShin3, 2018-06-04:
I've been trying to install the correct drivers for the TL-WN722N to work on my Rpi kali linux machine. wiki.kernel says that the wireless testing driver needs this firmware. lsusb won't list the proper Artheros chipset when called. I am just assuming that the dongle won't work without the proper drivers.

olerem, 2018-06-04:
I assume you have TL-WN722N v2, which is based on Realtek RTL8188EUS.
Is your usb id: 2357:010c?

SunTzuShin3, 2018-06-04:
That's correct, but currently no id comes up. i had another driver installed previously that when i called lsusb -v it would show Realtek, but in my infinite wisdom i thought 'that can't be correct, isn't it supposed to be Atheros?' and did a fresh install of kali linux and attempted this. 

https://github.com/mfruba/kernel/tree/master/drivers/TL-WN722N_v2.0-Ralink 
this was the first one that i tried

olerem, 2018-06-04:
Atheros is on version 1 of this adapter. Version 2 is Realtek.

SunTzuShin3, 2018-06-04:
sorry for wasting your time sir. can i erase this issue to hide my shame?

olerem, 2018-06-04:
Sure, no problem.

closed 2018-06-04 by SunTzuShin3