qca/open-ath9k-htc-firmware #148: CCA Threshold

jeter1112, 2018-07-27:
Hi, I wanted to change the CCA threshold of ar9271. 
I modified the ar9002_phy.h.

_#define AR_PHY_CCA_NOM_VAL_9271_2GHZ             -20
#define AR_PHY_CCA_MIN_GOOD_VAL_9271_2GHZ      -27
#define AR_PHY_CCA_MAX_GOOD_VAL_9271_2GHZ      -16_

However, it seems nothing to happen. I want to know how to change CCA threshold correctly.

erikarn, 2018-07-27:
Hi,

Thats for NF calibration. There's a separate CCA register which programs in
the noise floor value. It's done in the PHY init path.

-adrian

jeter1112, 2018-07-29:
Which register represents CCA register?

rodizio1, 2018-11-09:
@jeter1112 Not sure if it's still relevant for you, but here is a patch that allows for changing the threshold via module parameter:
https://github.com/bortek/EZ-WifiBroadcast/blob/develop/kernel/linux-4.9.28-ath9k_htc-misc.patch