qca/open-ath9k-htc-firmware #154: Mac Chip Rev 0x0f.3 is not supported by this driver

tjbp, 2018-11-28:
Consistently under all circumstances when using a AR9271 dongle on a Raspberry Pi 3 (ARM) device, I receive the following:

`[   70.318348] usb 1-1.2: new high-speed USB device number 6 using dwc2
[   70.469828] usb 1-1.2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[   70.478666] usb 1-1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   70.486647] usb 1-1.2: Product: USB2.0 WLAN
[   70.491097] usb 1-1.2: Manufacturer: ATHEROS
[   70.495620] usb 1-1.2: SerialNumber: 12345
[   70.501147] usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   70.801960] usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   71.054247] ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[   72.078419] ath: phy1: Mac Chip Rev 0x0f.3 is not supported by this driver
[   72.085728] ath: phy1: Unable to initialize hardware; initialization status: -95
[   72.093590] ath: phy1: Unable to initialize hardware; initialization status: -95
[   72.101486] ath9k_htc: Failed to initialize the device
[   72.109507] usb 1-1.2: ath9k_htc: USB layer deinitialized`

Using the identical firmware with the same donghle on an x86_64 machine it works fine:

`[ 7309.631285] usb 1-1.3: new high-speed USB device number 11 using ehci-pci
[ 7309.748793] usb 1-1.3: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[ 7309.748797] usb 1-1.3: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 7309.748800] usb 1-1.3: Product: USB2.0 WLAN
[ 7309.748802] usb 1-1.3: Manufacturer: ATHEROS
[ 7309.748805] usb 1-1.3: SerialNumber: 12345
[ 7309.749218] usb 1-1.3: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 7310.036469] usb 1-1.3: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[ 7310.288498] ath9k_htc 1-1.3:1.0: ath9k_htc: HTC initialized with 33 credits
[ 7310.552688] ath9k_htc 1-1.3:1.0: ath9k_htc: FW Version: 1.4
[ 7310.552692] ath9k_htc 1-1.3:1.0: FW RMW support: On`

#136 looks like the same issue but has since been closed. Interestingly it all describes hotplugging as working - I find nothing will work now, whether hotplugging or booting with the dongle attached.

olerem, 2018-11-28:
This issue is caused by dwc2 driver or hardware. dwc2 is cheap ip core with different issues. Most of the functionality is implemented in software and make it not so performant in comparison to the controller which you will usually found on a x86_64 machine.
One of huge issues of dwc2 is problem with mixed and fragmented transfers. Try to remove every thing else on the usb bus and test i again. In any case, it is not some what should be fixed by ath9k-htc driver or firmware

tjbp, 2018-11-28:
OK thanks, that's a fair enough explanation! Time to find a different platform...

closed 2018-11-28 by tjbp