qca/open-ath9k-htc-firmware #159: Crash in Ubuntu 20.04

tkircher, 2020-04-27:
I'm using the ALFA AWUS036NHA, Ubuntu 20.04 with kernel 5.4.0-26-generic x86_64

ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n

It takes a couple of tries to get it to recognize the device. It also eventually works. However, it repeats this error over and over in the system log every couple of seconds:

[  739.269145] ------------[ cut here ]------------
[  739.269149] usb 7-2: BOGUS urb xfer, pipe 1 != type 3
[  739.269174] WARNING: CPU: 3 PID: 2874 at drivers/usb/core/urb.c:478 usb_submit_urb+0x561/0x5e0
[  739.269177] Modules linked in: ccm nls_iso8859_1 nvidia_uvm(O) nvidia_drm(PO) nvidia_modeset(PO) nvidia(PO) uvcvideo snd_hda_codec_analog snd_hda_codec_generic snd_hda_codec_hdmi videobuf2_vmalloc ledtrig_audio videobuf2_memops ath9k_htc snd_usb_audio videobuf2_v4l2 ath9k_common ath9k_hw snd_hda_intel snd_intel_dspcfg snd_usbmidi_lib ath snd_hda_codec videobuf2_common mac80211 videodev snd_hda_core snd_hwdep snd_pcm cfg80211 drm_kms_helper libarc4 mc snd_seq_midi ipmi_devintf input_leds snd_seq_midi_event snd_rawmidi ipmi_msghandler snd_seq fb_sys_fops coretemp snd_seq_device snd_timer kvm_intel syscopyarea snd sysfillrect sysimgblt kvm soundcore asus_atk0110 serio_raw mxm_wmi mac_hid sch_fq_codel parport_pc ppdev lp parport drm ip_tables x_tables autofs4 hid_generic firewire_ohci pata_marvell psmouse i2c_i801 skge firewire_core ahci crc_itu_t pata_acpi lpc_ich sky2 libahci floppy wmi usbhid hid
[  739.269224] CPU: 3 PID: 2874 Comm: kworker/u8:0 Tainted: P        W  O      5.4.0-26-generic #30-Ubuntu
[  739.269226] Hardware name: System manufacturer P5Q DELUXE/P5Q DELUXE, BIOS 2301    07/10/2009
[  739.269232] Workqueue: phy10 ath9k_htc_ani_work [ath9k_htc]
[  739.269234] RIP: 0010:usb_submit_urb+0x561/0x5e0
[  739.269236] Code: 00 89 4d c4 44 89 55 c8 44 89 45 d4 e8 28 9d f0 ff 44 8b 45 d4 8b 4d c4 4c 89 fa 48 89 c6 48 c7 c7 48 4d 80 90 e8 1a 0d 8b ff <0f> 0b 44 8b 55 c8 e9 e9 fb ff ff 66 66 66 66 90 b8 a6 ff ff ff e9
[  739.269237] RSP: 0018:ffffacf70039bc68 EFLAGS: 00010286
[  739.269239] RAX: 0000000000000000 RBX: 0000000000000002 RCX: 0000000000000006
[  739.269240] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff9d8c77b978c0
[  739.269241] RBP: ffffacf70039bca8 R08: 00000000002523f5 R09: 0000000000000004
[  739.269242] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9d8c6b90a0c0
[  739.269244] R13: ffff9d8c6f040000 R14: 0000000000000002 R15: ffff9d8c71bad510
[  739.269247] FS:  0000000000000000(0000) GS:ffff9d8c77b80000(0000) knlGS:0000000000000000
[  739.269250] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  739.269251] CR2: 00007f95025d24e8 CR3: 000000022749e000 CR4: 00000000000006e0
[  739.269252] Call Trace:
[  739.269257]  hif_usb_send+0x189/0x320 [ath9k_htc]
[  739.269261]  htc_issue_send.constprop.0+0x5e/0x70 [ath9k_htc]
[  739.269265]  htc_send_epid+0x15/0x20 [ath9k_htc]
[  739.269268]  ath9k_wmi_cmd+0x11c/0x1c0 [ath9k_htc]
[  739.269272]  ath9k_regwrite+0xdc/0x100 [ath9k_htc]
[  739.269276]  ath_hw_cycle_counters_update+0x123/0x170 [ath]
[  739.269287]  ath9k_hw_ani_monitor+0x1d/0x1e0 [ath9k_hw]
[  739.269291]  ath9k_htc_ani_work+0x188/0x1a0 [ath9k_htc]
[  739.269297]  process_one_work+0x1eb/0x3b0
[  739.269299]  worker_thread+0x4d/0x400
[  739.269301]  kthread+0x104/0x140
[  739.269303]  ? process_one_work+0x3b0/0x3b0
[  739.269304]  ? kthread_park+0x90/0x90
[  739.269307]  ret_from_fork+0x35/0x40
[  739.269311] ---[ end trace 86796d46a0df18b2 ]---

Maybe someone knows what's happening here.

tkircher, 2020-04-27:
I wonder if this is related:

https://github.com/kuoruan/lede/blob/master/target/linux/generic/pending-4.9/821-usb-Remove-annoying-warning-about-bogus-URB.patch

olerem, 2020-04-28:
The patch is just removing warning. Removing sanity checks is not a solution.

Probably this is related: "It takes a couple of tries to get it to recognize the device."
Did you tried to use different cable? Something shorter?

olerem, 2020-04-28:
And, please visit this page: https://github.com/qca/open-ath9k-htc-firmware/wiki/Troubleshooting-and-bug-reporting

tkircher, 2020-04-28:
Thanks for the response. I dug out a shorter cable, and no errors. Closing.

closed 2020-04-28 by tkircher