qca/open-ath9k-htc-firmware #161: AR9271

Dimelay, 2020-07-14:
Hi!
after upgrade not working  AR9271

5.7.0-1-amd64 #1 SMP Debian 5.7.6-1 (2020-06-24) x86_64 GNU/Linux
cat /etc/debian_version 
bullseye/sid

[  505.033319] usb 1-1.2: new high-speed USB device number 7 using ehci-pci
[  505.158860] usb 1-1.2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[  505.158865] usb 1-1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  505.158867] usb 1-1.2: Product: USB2.0 WLAN
[  505.158869] usb 1-1.2: Manufacturer: ATHEROS
[  505.158871] usb 1-1.2: SerialNumber: 12345
[  505.159492] usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  505.159709] usb 1-1.2: firmware: direct-loading firmware ath9k_htc/htc_9271-1.4.0.fw
[  505.446738] usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  505.699065] ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[  617.272563] ath9k_htc: Failed to initialize the device
[  617.272665] usb 1-1.2: ath9k_htc: USB layer deinitialized

erikarn, 2020-07-14:
Hi!

The firmware hasn't changed in a long time. Has something changed on the
kernel USB support? Which kernel version does it work with?

-adrian

Dimelay, 2020-07-14:
works with 5.6.0-2-amd64 #1
I will try to roll back and check. Thank!

Xstoudi, 2020-08-06:
Same problem :
```
Linux xavier 5.3.0-64-generic #58-Ubuntu SMP Fri Jul 10 19:33:51 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

erikarn, 2020-08-06:
hi!

Please roll back to an earlier kernel version and re-test. It's possible
something at the USB layer changed and is unhappy with this chipset.

-a

shiftee, 2020-12-01:
This issue was almost certainly this one: https://bugzilla.kernel.org/show_bug.cgi?id=208251
The bug has been resolved, suggest closing this.

closed 2020-12-01 by erikarn