qca/open-ath9k-htc-firmware #165: CPU CLK  200MHz？

ItWillBeSkrskr, 2020-11-04:
Hi ！Sorry to bother you.  
I have some questions about this firmware.
I‘ve got the data sheet of ar9271 .  But I feel it doesn't match the source code to a certain extant.
For example，some registers' name  in the datasheet are different with this code .(And why there are source files prefixed with ar5416?)  
Maybe because I'm a new hand , I didn't find something can tell me the background about this program after browsing the doc and wiki.

So，it is the most critical problem that I can't handle on my own  :
[image]
[image]

I want to set the CPU clk to 40MHz but in your code it has been set to 200MHz.
[image]

And I can't find the regitsters like 0x0005_6000 in the datasheet .
[image]

olerem, 2020-11-04:
This firmware runs on two different SoCs:
- ar9271 - code name k2
- ar7010 - code name magpie

The reason why this firmware has artifacts or names  pointing to other SoCs is following:
- most company prefer to save resources so they develop one firmware/driver for many chips
- this FW is historically grown from BSD source code

ItWillBeSkrskr, 2020-11-04:
Thank you so much!
Maybe these convention and  knowledge are ordinary to you firmware developers . BUT it is really important to us who just want to modified the firmware a little bit.

ItWillBeSkrskr, 2020-11-04:
There are something I got still puzzled with:

As the function **_change_magpie_clk()_** in **_athos_main.c_**  I mentioned above ,  I want to know  if it will really work with chip ar9271  ,  or I need to add some code with functions in k2 project instead of magpie ?

olerem, 2020-11-04:
> I want to know if it will really work with chip ar9271
no

> I need to add some code with functions in k2 directory instead of magpie

sboot is not part of compiled firmware. It is a reference code of boot loader.

You will probably wont to do it somewhere in target_firmware/magpie_fw_dev/target/init/app_start.c or target_firmware/magpie_fw_dev/target/init/init.c

rodrigowue, 2020-12-02:
Hey!
Pretty interesting discussion. I'm not sure if I got it right. So, if we want to change the frequency on the cpu we need to manipulate the sboot? @Skrrrrrrrrrrrr do you have any updates on this?

Cheers   :)

Skrrrrrrrrrrrr, 2020-12-08:
> Hey!
> Pretty interesting discussion. I'm not sure if I got it right. So, if we want to change the frequency on the cpu we need to manipulate the sboot? @Skrrrrrrrrrrrr do you have any updates on this?
> 
> Cheers :)

Sorry to reply to you not on time.

In fact ,  I'm going to  modified this NIC to add some features.
So it is necessary to figure out  how to write registers to control the workflow in the device.
The modification of CPU_CLK is just a piece of the whole plan.

Skrrrrrrrrrrrr, 2020-12-08:
@rodrigowue

rodrigowue, 2020-12-09:
Thanks a lot!