qca/open-ath9k-htc-firmware #166: ACK handler

matus1604, 2020-11-14:
Hi guys,
I would really appreciate your help. I want to modify the firmware, maybe the driver too to be able to create virtual access points for individual devices. First of all, I'm looking for the part of the code which handles sending ACK according to BSSID. I've already gone through the driver, but I didn't find anything useful, so I guess, it's gonna be somewhere in the firmware.

Don't you know which part of the code handles sending ACK according to BSSID addresses?

Thanks for your help

erikarn, 2020-11-14:
hi!

So, it's the same as the other atheros devices of the era. If it matches
the BSSID/BSSMASK registers the hardware will generate ACKs. The RTS, CTS
responses ACKs are generated by hardware, not by firmware/software.

matus1604, 2020-11-15:
Thank you for your fast reply

Is it possible to somehow change BSSID registers or this HW is completely separated from firmware and whole SW?

erikarn, 2020-11-15:
hi!

So, there are two variants:

* AR9280+AR7010 - that's the AR9280
* AR9271 - that's the AR9285 + AR7010 on the same die

I don't remember off hand if the firmware or the host driver controls the
BSSID/BSSMASK registers on ath9k_htc, but it should be pretty easy to grep
for BSSID in ath9k and the ath9k htc firmware code to see what's up.

-adrian

matus1604, 2020-11-30:
Hi,
thank you very much for the info, it was really helpful, I've finally get somewhere after some digging and greping. 

Firstly, I have TL-WN722N adapter, so it's AR9271 chip(ath9k_htc driver).

So, I found function setbssidmask() in the driver code, structure "common" is an argument, where I can change the bssidmask (common->bssidmask). I managed to reuse patch for ODIN (to change bssid mask) (https://github.com/josemariasaldana/odin-utilities/tree/master/ath9k_htc), so now I am able to change bssid mask anytime with debugfs.

I tried this scenario:

Phone MAC: 94:65:2d:38:dc:ef
AP MAC: 14:cc:20:1b:6f:66

Manually computed bssid mask:
7f:56:f2:dc:4c:76

I compiled, reloaded driver, run AP mode, changed bssid mask from default FF:FF:FF:FF:FF:FF to mine 7f:56:f2:dc:4c:76, checked DMESG for debug message from the driver, if it is changed, everything was OK, phone and AP could communicate(AP was sending ACK for phone requests), but when I changed bssid mask for anything else, even all zeros, nothing changed, AP was still sending ACK to phone, but it shouldn't do that.

This is the reused code from ODIN patch for ath9k_htc, when I change file in debugfs with new bssid mask, it run this function "write_file_bssid_extra"

`static ssize_t write_file_bssid_extra(struct file *file, const char __user *user_buf,
                size_t count, loff_t *ppos)
{
        struct ath9k_htc_priv *priv = file->private_data;
        struct ath_common *common = ath9k_hw_common(priv->ah);

        char buf[32];
        u8 macaddr[ETH_ALEN];
        ssize_t len;

        len = min(count, sizeof(buf) - 1);
        if (copy_from_user(buf, user_buf, len))
                return -EFAULT;

        buf[len] = '\0';

        sscanf(buf, "%hhx:%hhx:%hhx:%hhx:%hhx:%hhx", &macaddr[0], &macaddr[1],
                        &macaddr[2], &macaddr[3], &macaddr[4], &macaddr[5]);

        memcpy(common->bssidmask, macaddr, ETH_ALEN);
        printk(KERN_DEBUG "BSSID MASK change\n");
        ath_hw_setbssidmask(common);
`

Do you know what I could be possibly doing wrong? I want to change bssid mask for example to 00:00:00:00:00:00 to stop communication between phone and AP, after that, change it back to 7f:56:f2:dc:4c:76 to start it again (sending ACK)

I appreciate your help so much.

erikarn, 2020-11-30:
hi,

Since RTS, CTS, ACK etc are generated by the hardware and not by the
firmware, I don't know if you're going to be able to implement the feature
this way. You don't get enough signaling to the driver to be able to
manually flip the bssmask between receiving a data frame AND before it
sends the ACK response.

-adrian