qca/open-ath9k-htc-firmware #173: random reset and complete lock

varoudis, 2022-02-03:
Hi,

At random times and tested with two different Linux distro running 5.10 and 5.15 the usb device goes dead until is physically removed from the system.
Everything seems ok after USB in re-inserted until a random interval later same thing happens again.

Any ideas?
USB card is ALFA AWUS036NHA

```
		[    5.171311] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
		[    5.173528] usbcore: registered new interface driver ath9k_htc
		[    5.175441] kmodloader: done loading kernel modules from /etc/modules.d/*
		[    5.469340] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
		[    5.751851] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
		[    6.555202] 8021q: adding VLAN 0 to HW filter on device eth0
		[    6.557432] br-lan: port 1(eth0) entered blocking state
		[    6.559144] br-lan: port 1(eth0) entered disabled state
		[    6.561547] device eth0 entered promiscuous mode
		[    6.564435] br-lan: port 1(eth0) entered blocking state
		[    6.566314] br-lan: port 1(eth0) entered forwarding state
		[    6.934851] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
		[    6.936906] ath9k_htc 1-1:1.0: FW RMW support: On
		[    6.938809] ath: EEPROM regdomain: 0x833a
		[    6.940534] ath: EEPROM indicates we should expect a country code
		[    6.943417] ath: doing EEPROM country->regdmn map search
		[    6.945221] ath: country maps to regdmn code: 0x37
		[    6.946591] ath: Country alpha2 being used: GB
		[    6.947894] ath: Regpair used: 0x37
		[    6.980456] ieee80211 phy0: Atheros AR9271 Rev:1
		[    8.452273] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
		[    8.496708] device br-lan entered promiscuous mode
		[  127.581219] wlan0: authenticate with 00:2a:10:cf
		[  129.527475] wlan0: send auth to 00:2a:10:cf (try 1/3)
		[  129.533347] wlan0: authenticated
		[  129.539989] wlan0: associate with 00:2a:10:cf (try 1/3)
		[  129.545750] wlan0: RX AssocResp from 00:2a:10:cf (capab=0x431 status=0 aid=12)
		[  129.615347] wlan0: associated
		[  129.723574] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
		[ 2571.637692] ath: phy0: Unable to remove station entry for: 00:2a:10:cf
		[ 3038.032661] usb 1-1: USB disconnect, device number 2
		[ 3038.125724] ath: phy0: Chip reset failed
		[ 3038.126858] ath: phy0: Unable to reset channel (2447 Mhz) reset status -22
		[ 3038.128678] ath: phy0: Unable to set channel
		[ 3038.230731] ath: phy0: RX failed to go idle in 10 ms RXSM=0xc0783b01
		[ 3038.434341] ath: phy0: RX failed to go idle in 10 ms RXSM=0x0
		[ 3038.446391] ath: phy0: Failed to wakeup in 500us
		[ 3038.548547] ath: phy0: RX failed to go idle in 10 ms RXSM=0x4016cf01
		[ 3038.752421] ath: phy0: RX failed to go idle in 10 ms RXSM=0xc0683b01
		[ 3038.764229] ath: phy0: Failed to wakeup in 500us
		[ 3038.995863] ath: phy0: RX failed to go idle in 10 ms RXSM=0xc0683b01
```

olerem, 2022-02-03:
Hi,

please take a look here:
https://github.com/qca/open-ath9k-htc-firmware/wiki/usb-related-issues

varoudis, 2022-02-03:
@olerem Thanks for the quick reply. I did check and for about an hour Im running a test with:

options "nohwcrypt=1 ps_enable=0 blink=0"

I think blink=0 is the same as the "sudo bash -c "echo none > /sys/class/leds/ath9k_htc-phy*/trigger"" ?
LED isnt blinking anymore

olerem, 2022-02-03:
yes. Will be interesting to know results.

olerem, 2022-02-08:
Sorry, Currently I have no idea. To get more information, you will need to get access to the UART pins (TXD1, RXD1) on the board of this adapter (baudrate 19200) and log this error on the firmware side.

varoudis, 2022-02-08:
ok Ill try to find some time to do it. (and find my ftdi adapter first :) )

ZerBea, 2022-04-01:
I got some issue reports possible regarding this issue, too, and it looks like they don't appear on kernel 5.17.1 any longer.
But that need some more investigations. The last test over a period of nearly 9h is looking good, so far:
https://github.com/ZerBea/hcxdumptool/issues/209#issuecomment-1085460482