qca/open-ath9k-htc-firmware #174: ath9k doesn't stop freezing several times in an hour

lutzig, 2022-05-03:
Good morning !

several times in an hour I have to reactivate the Wi-Fi cause the speed falls to 1 Mbt/s.
I've done all checks concerning the Livebox - everything is OK.

@HP-14-TS:/$ uname -a
Linux HP-14-TS 5.15.0-27-generic #28-Ubuntu SMP Thu Apr 14 04:55:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
@HP-14-TS:/$ 

@HP-14-TS:/$ lspci | grep -i net
08:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL810xE PCI Express Fast Ethernet controller (rev 07)
0a:00.0 Network controller: Qualcomm Atheros AR9485 Wireless Network Adapter (rev 01)
@HP-14-TS:/$ 

@HP-14-TS:/$ sudo rfkill list
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
@HP-14-TS:/$ 

@HP-14-TS:/$ sudo iw dev wlo1 link
Connected to a4:3e:51:ed:20:b9 (on wlo1)
	SSID: Livebox-20B9-24
	freq: 2427
	RX: 80638 bytes (798 packets)
	TX: 16912 bytes (99 packets)
	signal: -59 dBm
	rx bitrate: 54.0 MBit/s
	tx bitrate: 54.0 MBit/s

	bss flags:	short-preamble short-slot-time
	dtim period:	3
	beacon int:	100
@HP-14-TS:/$ 

@HP-14-TS:/$ iw dev wlo1 station dump
Station a4:3e:51:ed:20:b9 (on wlo1)
	inactive time:	1144 ms
	rx bytes:	296074
	rx packets:	2751
	tx bytes:	71199
	tx packets:	360
	tx retries:	10
	tx failed:	0
	beacon loss:	915
	beacon rx:	1185
	rx drop misc:	19
	signal:  	-60 [-60] dBm
	signal avg:	-59 [-59] dBm
	beacon signal avg:	-59 dBm
	tx bitrate:	54.0 MBit/s
	tx duration:	32302 us
	rx bitrate:	54.0 MBit/s
	rx duration:	51357 us
	airtime weight: 256
	expected throughput:	18.584Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	3
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	121 seconds
	associated at [boottime]:	20486.293s
	associated at:	1651568792860 ms
	current time:	1651568914177 ms
@HP-14-TS:/$ 

Thank you in advance for your help !
Regards,

olerem, 2022-05-03:
hi @lutzig . This repository for ath9k-htc firmware, which is used on ar7010 and ar9271 USB WiFi controller.  Your question seems to be related to AR9485 PCI WiFi controller, which is not using ath9k-htc firmware and not using ath9k-htc kernel driver.

lutzig, 2022-05-03:
Hi Olerem!
Do you want to mean that there is here nobody who could help me ?
Do you know where I could find some help, please ? Since the upgrade to UBUNTU 22.04 I am looking for it and I find nowhere else that here :o... And that's getting on my nerves...
Thank in advance.